gnea/grbl #523: Homing pulloff

sfinexer, 2018-09-30:
I think that the homing_pulloff value for each axis should be set separately.
case 27: settings.homing_pulloff = value; break;
This is important if the origin is not equidistant from the endpoints.

neilferreri, 2018-09-30:
Your work origin is set independently of machine origin. The homing process is just to have a reference location for positioning.

sfinexer, 2018-09-30:
@neilferreri There are limit switches which have a large hood, and who have a small stroke.
And for different types of switches, you need a different homing_pulloff.

sfinexer, 2018-09-30:
For each coordinate, you need your own homing_pulloff value.

sfinexer, 2018-09-30:
In other words, homing_pulloff or not enough to open the end sensor for one axis, or more than moving on the other axis.

MARIOBASZ, 2018-09-30:
.... place 5 mm on all axes !!!

sfinexer, 2018-09-30:
@MARIOBASZ 
I need separately along the axes.
I want to immediately get the start point of the workpiece.
And the workpiece is in coordinates X40 Y5. if there are different parameters, I immediately position the machine in the desired position with the zero search cycle.

MARIOBASZ, 2018-09-30:
G53 is the position in machine coordinates.
G54 onwards, work positions. You locate the current work position 0,0,0 in any position of machine coordinates (example the machine coordinate -20,12, -5 can be 0,0,0 for G54 and 13,5, 21,0, 12,342 for G55)

sfinexer, 2018-09-30:
> .... place 5 mm on all axes !!!
5mm small for X axis.

> but it is too small for the X axis to disable the end sensor.

MARIOBASZ, 2018-09-30:
End stop in the end of axe

closed 2018-09-30 by sfinexer