gnea/grbl #524: Plasma cutter touch-off cycle

infernozx, 2018-09-30:
I have built a CNC plasma cutter from scratch using GRBL, and an arduino.

I'm using Universal G-Code Sender to control the GRBL, and one of the features that I thought I could get away with not having before, is really starting to become an issue.

Many CNC plasma cutters, in order to control the height of the torch above the work piece will run a touch-off subroutine. This compensates for a lot of the warping/shifting due to heat application.

The following is an excerpt from my code:

`N0120 Z3.2
N0130 F100 G38.2 Z-10
N0140 F20 G38.4 Z10
N0150 G92 Z0
N0160 F50 Z3.20
N0170 M03`

From my understanding:

F100 (feed speed) G38.2 (probe toward work piece stop on contact) Z-10 (direction towards work piece, and a point to stop and error if no contact is met)

This piece of code seems to work. It probes down, and stops when the torch contacts, and the switch is tripped.

Then the next line is supposed to withdraw the probe, and stop on loss of contact (G38.4). It does, but as soon as the contact is lost, it throws a "ALARM: Probe Error"

Ideally, as soon as it stops, I want it to move onto the G92, which is setting the Z height of the top of the plate at 0, then moving up to my 3.2mm cut height before firing the torch and starting with the cut.

So I think I've localised my problem to the G38.4 command, but I don't understand why it doesn't work, as the 38.2 seems to work fine. 

Any advice? I know this may not be a strictly GRBL problem, but you guys seem to know the most about the finiky bits of G-Code and how grbl is interpreting the commands.

Or.. am I going about this all the wrong way?

langwadt, 2018-10-01:
intermittent contact so the probe already lost contact before g38.4 ?

infernozx, 2018-10-01:
Im pretty sure this is the answer. 

I got frustrated after fighting this all day and took the whole probe system apart, laid it out on the bench and tested it. Definitely intermittent. 

Waiting on parts to put it back together and then I'll try it again. 

I imagine the 38.2 followed by the 38.4 is an acceptable way of completing this operation?

closed 2018-10-01 by infernozx