gnea/grbl #525: Upgrading GRBL on 3018/2418/1610 CNC router

lawolff, 2018-09-30:
I wonder if I could get anyone to tell me how to flash this firmware to v1.1. I can't seem to do it directly with Arduino IDE, apparently you've got to compile it and that process is in the wind. I don't see a way to use the binary file (.hex) directly to flash.

biasedlogic, 2018-10-02:
If your board does not respond as Arduino (mine did, Arduino nano bootloader), then you need to 'cold' flash it over ISP interface just like any other Atmel chip. For that you'd use the hex file...

cri-s, 2018-10-02:
You can use this  http://russemotto.com/xloader/ to flash .hex file in
arduino boards without the need to Compiler it Form source.

BR
Cri

JBSchueler, 2018-10-09:
I have updated my CNC 3018 to GRBL 1.1 via USB.
Please read issue #336 ([Update Woodpecker CNC GRBL 0.9 to 1.1f](https://github.com/gnea/grbl/issues/336))

itchygrub, 2019-07-01:
JBSchueler
In simple terms how did you do it

JBSchueler, 2019-07-01:
https://github.com/gnea/grbl/issues/336#issuecomment-372070897

swellhunter, 2019-07-23:
also on [ #336](https://github.com/gnea/grbl/issues/336#issuecomment-372070897) 

As an avrdude switch the -B100 cures a lot of ills, at very little cost of speed.  I'd leave it in the makefile.

I'd also be inclined to swap in an FTDI uno or nano.

If you are not a Mac user, and especially if you run Linux,  you might look into flashing with the AVR plugin from Eclipse. You are still going to need a programmer. Otherwise go back to the makefile and resolve the avrdude command line and paste in -B100 as seen in [#336](https://github.com/gnea/grbl/issues/336#issuecomment-372070897) above. 90% of the hobbyist population uses USBAsp or USBTiny as a programmer and need to do this sooner or later for bad signatures and low speed oscillator clocking.

The fact that we flash fuses and are clobbering the bootloader means that it is not a stock Arduino any more in any case, it is just a carrier board for an ATMega328P.  I note that a lot of "nano" 3018's ship without the ICSP pins, one would also hope they are not glued in by the other pins.  

A really good project would be a 3D rendered simulator harness to test this.