gnea/grbl #529: Homing issue. $H homes X or Y

simplesimson, 2018-10-02:
$H homes only X or Y which ever axis does hit the endstop first.
Any suggestions? 
My $23 is currently set to 3 ( it shall home in the front left and Z up.) So Y negative X Negative Z up
I´m using the universal gcode sender and GRBLv1.1

My Homing cycle **shall** look like:
_____________________
|         x                   |
|                              |
|                              |
|                              |
|                              |
_____________________ Zmoves up.
^ here is the target

_____________________
|                              |
|                              |
|                              |
|                              |
|        x                     |
_____________________ Y homes

_____________________
|                              |
|                              |
|                              |
|                              |
| x                            |
_____________________ x homes

But At the moment it looks like this.
_____________________
|         x                   |
|                              |
|                              |
|                              |
|                              |
_____________________ Zmoves up.

Now X and Y start moving

_____________________
|       x                      |
|    x                      |
|  x                           |
|x                             |
|                             |
_____________________  X Homes before Y. Y doesnt start moving again.

So how to deactivate the movement of the two axis at the same time?

[Here ](https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle)you can read this
"X and Y axis
X, Y Axis move both to Homing direction at fast rate ($25)
The first Axis triggers the switch will stop and wait for the second axis to trigger"

Thats just not the case. Both move at the same time. 

Thank you for your time and the awesome software!

simplesimson, 2018-10-02:
Thanks for your answer.
I did the config with the GUI of estlcam so i don´t have an config file :( maybe i just have to redo it all by hand.

chamnit, 2018-10-02:
@simplesimson : No, this isn't part of the Grbl configuration. You have to edit Grbl's source code, re-compile, and flash Grbl onto your Arduino, if you want to alter the homing cycle behavior.

simplesimson, 2018-10-02:
Yes. I got what you mean. In Estlcam there is a feature to do all the cfg things with a gui.
[image]

simplesimson, 2018-10-02:
So I actually never touched the cfg file before uploading because i was sure that estlcam will handle it all. It kind of did. While using estlcam it all worked fine. But estlcams preview of the gcode ist horrible so i switched to UGS

109JB, 2018-10-03:
You are still wrong. That page from estlecam or whatever is only the EEPROM settings. Those settings can be configured after compiling and loading grbl to an arduino.

There is a config.h file that has compile-time options. These options have to be changed in the config.h file and then grbl compiled and flashed to the arduino. The homing behavior is edited in the config.h file.

simplesimson, 2018-10-04:
Well it turned out that the problem was a lose cable on the Arduino powering up the 5V of my optocouplerboard. I replaced it and glued the new one inplace. Now its all working again. 

And yes, you might be right that estlcam only changes EEPROM. But i stoped using it anyways as it interprets gcode wrong.

chamnit, 2018-10-04:
@simplesimson : Thanks for reporting.

closed 2018-10-04 by chamnit