gnea/grbl #531: how to correctly calibrate PWM for laser QUESTION

sinfocomp, 2018-10-06:
Hello, i have being running GRBL for more than a year now and my laser responded pretty good to the power level acordingly to my setting, i use $30 = 100 and $31 = 0, this responded ok,
however i have changed my psu becouse it burned and the new one responds diferently
i have to use S53 where i used S21 before to get same result, and from S55 upwards i get 100% power
so i cannot correctly control laser power, 
how can i fix this? is there a setting i should change to get better control ?
please advice
Ed.

chamnit, 2018-10-06:
There is a nonlinear spindle option for laser output in particular. Look near the bottom of the config.h file on how to create and enter the spindle profile.

sinfocomp, 2018-10-06:
This is all i could find regarding spindle PWM, please forgive my poor understanding on this but isnt this just like changeing the $30 and $31 settings ?

#define VARIABLE_SPINDLE // Default enabled. Comment to disable.

// Used by variable spindle output only. This forces the PWM output to a minimum duty cycle when enabled.
// The PWM pin will still read 0V when the spindle is disabled. Most users will not need this option, but
// it may be useful in certain scenarios. This minimum PWM settings coincides with the spindle rpm minimum
// setting, like rpm max to max PWM. This is handy if you need a larger voltage difference between 0V disabled
// and the voltage set by the minimum PWM for minimum rpm. This difference is 0.02V per PWM value. So, when
// minimum PWM is at 1, only 0.02 volts separate enabled and disabled. At PWM 5, this would be 0.1V. Keep
// in mind that you will begin to lose PWM resolution with increased minimum PWM values, since you have less
// and less range over the total 255 PWM levels to signal different spindle speeds.
// NOTE: Compute duty cycle at the minimum PWM by this equation: (% duty cycle)=(SPINDLE_PWM_MIN_VALUE/255)*100
// #define SPINDLE_PWM_MIN_VALUE 5 // Default disabled. Uncomment to enable. Must be greater than zero. Integer (1-255).

chamnit, 2018-10-07:
Look at the current version and there are references for a piecewise linear fit.