gnea/grbl #532: Galvo

SerkanGuryalcin, 2018-10-07:
Hello dear I'd like to do galvolaser and ı need to read gcode. My project need 2 axis x and y so how can ı use grbl

alflexRH, 2018-10-10:
the simple answer is: you cant. GRBL can only do steppers, not galvos.

SerkanGuryalcin, 2019-01-20:
I see. Do you know how can ı read gcode for galvo motors ?

langwadt, 2019-01-20:
if there is room for the code you could just modify GRBL, leave it mostly as is but instead of step/direction output, just have the galvos follow the resulting position

X3msnake, 2019-01-21:
> yes, but usually when people talk about servos they mean 3-phase BLDC-type
> servos, those that can do multiple revolutions, and their controllers take
> step/direction (or other) pulses.
>
> or the 1-2ms pulse RC hobby servos).

so galvos are more like fixed motion rc servos but driven by voltage...

SerkanGuryalcin, 2019-01-21:
So can I read g code with galvos ??

X3msnake, 2019-01-21:
I think you would need to change the way GRBL sends things to your galvo
drivers.

We are talking of going from a looping model where each full tur
corresponds to a number of steps that then correspond to a given axel
motion to something that is more like controlling a PWM laser from 0-255
where 0 is the minimum position of your set table boundary and 255 is the
max.

I don't know enough programming to know what you would have to change or
where but i think it might not be as easy as swapping a few lines of code,
I may be wrong tho.

21/01/2019 à(s) 17:48:

X3msnake, 2019-01-21:
some google Fu returned this:

----
http://www.buildlog.net/blog/2018/02/laser-galvo-control-experiments/

----
https://photonlexicon.com/forums/archive/index.php/t-27033.html

"According to the specification, all X / Y / Synch control signals should be synchronized with the 2MHz clock signal. In Arduino UNO it is very difficult. " 

----
https://plus.google.com/communities/109609533907401814291/stream/401252fa-796c-4680-afa6-961794b39ac8

[image]

https://github.com/Blacksheep70/Marlin-OpenSLA

SerkanGuryalcin, 2019-01-21:
Thanks a lot I will try all methods

X3msnake, 2019-01-21:
@SerkanGuryalcin 
Keep us posted here on your results i'm curious about this too :)

TehseenHasan, 2021-01-01:
What if I make **Stepper Motors based Galvo**  then control it with GRBL.? then is it possible and effective.?
For example like this guy done:  https://www.youtube.com/watch?v=46jKSFHXgiI