gnea/grbl #533: Spindle Enable Stays High

lnelson214, 2018-10-07:
I've done a lot of searching on this and other forums but can't seem to track down my answer.

I'm using a generic CNC shield atop a generic UNO loaded with GRBL 1.1, with bCNC on a raspberry pi functioning as the host.

Under default settings, with variable spindle enabled, the voltage at the Spindle Enable pin stays at 5v no matter what I do. If I disable the variable spindle, spindle enable now works like it's supposed to, but I can't take advantage of the TTL functionality of the laser diode.  The laser that I'm using is an Eleksmaker LA03-3500 3.5W laser with TTL. I have the Power +/- connected to a 12v p/s and connecting the TTL wires to spindle enable (again, with variable spindle disabled) is the only way I can get any use out of it.

I know that disabling the spindle causes pins 11 and 12 on the UNO to swap, so my first thought was to swap the TTL to the Z-limit pins and enable the variable spindle, but those pins output 0v no matter what I'm doing with the spindle in bCNC. I've tried changing the max spindle rpm to 255, but it doesn't seem to change anything. I've also tried messing with the USE_SPINDLE_DIR_AS_ENABLE_PIN settings in the grbl firmware but it also doesn't change anything. The spindle enable pin always reads 5v unless I disable the variable spindle.

Is there something I'm doing wrong? Is there a setting, either in the grbl firmware or in the bCNC config that I'm overlooking? I assume that since my laser diode has TTL functionality, and it's something that's supported in grbl 1.1, it should work, no?

I've learned a lot from my research and have been able to systematically solve all the problems I've encountered thus far, but this one has got me stumped. I would really like to be able to do grayscale engraving with this machine, but without the TTL that's not possible right?

Thanks for any help you guys can offer on this matter, I really appreciate it.

Schildkroet, 2018-10-07:
If the spindle is enabled, it stays enabled. Theres no way to disable it, except a reset.
I don't know, if this is a bug or a feature

lnelson214, 2018-10-07:
Not sure what you mean. The spindle enable pin reads 5v as soon as I turn it on. I don't have to turn it on or pass any gcodes to grbl for it to be 5v.

chamnit, 2018-10-07:
With variable spindle enabled, you need to enable the spindle with M3 and set a non zero spindle speed. You’ll then see the pin voltage increase.

lnelson214, 2018-10-08:
> With variable spindle enabled, you need to enable the spindle with M3 and set a non zero spindle speed. You’ll then see the pin voltage increase.

I tried that but it has no effect on the voltage at the pin. If I have variable spindle enabled (so that I can use the M3 command with a speed, the pin voltage stays at 5v no matter what speed I set it to. There is no change. If I have it set to disable variable spindle, The speed command doesn't do anything, only M3 and M5 turns laser full on/full off.

chamnit, 2018-10-08:
With variable spindle enabled, the spindle pin is on D11. Disabled, it’s on pin D12.

lnelson214, 2018-10-08:
Yes, but when I hook the laser TTL up to D12, it doesn't work. The laser never powers on. If I check the voltage on D12 it reads zero no matter what speed I set the spindle. I have the GRBL configured to use 0 to 255 as the min/max spindle rpm when the variable spindle is enabled, but I've tried it with values up to 18000. Nothing.

lnelson214, 2018-10-08:
Maybe I have the wrong pin. The information I've found says that on the CNC shield the Z- limit pin goes to D12, but I've also seen it as Z+ some places. Either way, I've tried them both and it doesn't seem to make any difference.

chamnit, 2018-10-08:
If you have laser mode enabled, it behave different. See the wiki laser page as to why. You have to move with a G1 move to turn on the laser.

lnelson214, 2018-10-08:
That's the bit I missed. I think when I was trying it the program was using G0 codes but I'll have to double check when I get back. Will post an update when I figure out if it works. Thanks!

lnelson214, 2018-10-08:
Ok that was definitely it. I guess I just missed that part about the codes changing. I was using Inkscape to generate my Gcode, using the raster2gcode extension, but it didn't have any S codes in it. I assumed that even still I should be able to turn on/off the laser using the bCNC controls. Anyway, I'm now running a program successfully and everything is looking great. 

Another question, however. This program I created using laserGRBL on my windows laptop. I was wondering what are some good solutions for generating the files on a raspberry pi running raspbian? I've read a little bit about t2laser, but was wondering what others may be out there, and whether you have any favorites?

Anyway thanks for the help, I really appreciate it!

chepo92, 2018-11-09:
You can send $$ and look for $32 enabled

> 
> 
> an associated question, is there any hardware p8n that can tell me when grbl is in laser mode?

thawkins, 2018-11-09:
thats not really what i was looking for, i want to lock out the laser hardware when not in laser mode,  i have a laser/cnc rig and it effectivly uses the spindle controls to drive the laser. im designing a new shield for mid range CNC systems.

chamnit, 2018-11-12:
@thawkins : At the moment, it's only the settings. You can modify the source code to drive a pin high or low when Grbl loads its settings.

closed 2018-11-12 by chamnit