gnea/grbl #534: GRBL Panel Support

1Metalguru, 2018-10-07:
Sorry, this is not really a GRBL issue, but I didn't know where else to post it.  

I was just on the GRBL Panel Wiki page and it seems Gerrit Visser is hanging up his spurs and ending support for the program with this last release.

This is a call to anyone who may want to take up the reigns and continue supporting and improving GRBL Panel in the future.  I think this is a great program, and doesn't deserve to die alone and unloved.

Contact Gerrit if you are interested.

StuartB4, 2018-10-26:
Why do people make programs that are unusable unless you know how to compile them.
What a lot of messing about.
Why not just upload a compiled version.

1Metalguru, 2018-10-26:
There is a compiled zip file on Github.

Regards,
Darrell Norquay - Calgary, Canada
3D Tech Works - 3DTechWorks.ca
403 701 2005 - ndarrell@telus.net

1Metalguru, 2018-10-26:
https://github.com/gerritv/Grbl-Panel/releases

Regards,
Darrell Norquay - Calgary, Canada
3D Tech Works - 3DTechWorks.ca
403 701 2005 - ndarrell@telus.net

StuartB4, 2018-10-26:
Ok. Thanks very much 1Metalguru.

MeJasonT, 2018-11-19:
Unfortunately i'm starting to look at Candle as an alternative but its not as feature rich as Gerrits.
I don't like candle for only having one set of work coordinates, no coolant control, although it does work via code.  Reading through the issues it looks like they pay very little attention to snagging any errors with it.

carneeki, 2018-11-19:
@MeJasonT have you ruled out Universal G-code Sender? 

It has a bunch of user-configurable buttons that you can put in the interface to do things like coolant, or change WCS. Written in Java (compiles with Maven), but Will Winder (the author) also puts regular .jar files out ready to run on Windows/Apple/Linux. Also runs on something as small as a Pi and works nicely with GRBL. The toolpath visualizer on Pi is about the only ongoing issue I have with it, but it works on full desktops fine.

MeJasonT, 2018-12-19:
I have looked at UGS but i'm a fantasist who wants the GUI to look something like a real panel.  I'm not saying it doesn't - its actually come on leaps and bounds.  I avoided Mach3 as it looks like the panel was drawn by a three year old.
My latest bug bear with grbl panel is the macro controls  are disabled during program run time, I tried to  write a pause and resume command to two buttons.  Writing the commands in code works ish.  unfortunately with the code reading ahead of the operation, the moment it sees a pause in the buffer it immediately holds and does not run commands leading up to the pause instruction.
I would like to get to a point where i can pause the code to run a macro which performs a tool change before resuming.  I keep finding and trying different gui's and one day i will find one im happy to stick with - grbl panel was the one that inspired me to start and i've been very happy with the functionality so far.  lets face it the professional panels have limitations, i have yet to find one that make tea.

wlaw, 2018-12-19:
Hello Jason,

I am a big Grbl Panel fan and have used it for a long time.

I use a macro that moves my spindle over to the home spot and pauses, then
I change the tool and press continue.
Once this is done, the machine lowers the tool on a slide that checks the
height of the tool (tool offset) then I can run the next program.
https://www.youtube.com/watch?v=dRl4OHCeuMo&t=5s

This requires me to create a 1 job per end mill that will be used, but it
works well.

I suspect that you could create a tool change G-Code for the post processor
so when a tool change is called you can:
1. Move the spindle up
2. Shut the spindle off.
3. Move it to a convenient location.
4. Pause the movement: M0
5. Once the tool is replaced, press continue on the panel or with the pin
on the controller.
The machine should continue from that point, just be sure the height
(offset) for the tool is correct.

MeJasonT, 2018-12-20:
Fantastic wlaw, thank you very much.
could you tell me how to call a macro in the gcode when the program is running, or is there anyway of calling one of the two grbl N# functions as an alternative perhaps.

wlaw, 2018-12-20:
I am not sure what program you are using to make your project file.
But I think you should be able to have the post processor insert the macro
code into the G-Code when you make a tool change call.

For instance, this is for Vectric - post processor development.
TOOLCHANGE
 +---------------------------------------------------
+  Commands output at toolchange
 +---------------------------------------------------
begin TOOLCHANGE “Commands”
Commands that are output when a change of tool is required.
 Variables and commands that might be used include:
Previous Tool Number = [TP]
Tool Number = [T]
Tool name =  [TOOLNAME]
Toolpath Name = [TOOLPATH_NAME]
Toolpath Pathname = [PATHNAME]
Toolpath File Name  = [TP_FILENAME]
Toolpath File Directory = [TP_DIR]
Toolpath Extension      = [TP_EXT]
Spindle Speed = [S]   R.P.M.
M3   M Code often used to turn spindle on (Clockwise rotation).
M5   M Code often used to turn spindle off.

So my thought is that by editing your post processor for your program, you
can have the "Macro" G-Code inserted into the program when ever you call a
tool change.
This might be:
G53 Z?? raise the spindle to a good height.
M05  to shut of the spindle
G53 X?? Y?? : to move to a good location
M0 to pause the program for the manual tool change. (press continue when
finished)
M03 to start the spindle again
Then the G-code should take over again moving to a location and start
cutting.

I have never done this but I think is should be possible.
Maybe someone with some post processor development can help out.

MeJasonT, 2018-12-20:
I'm using Cambam with a grbl postprocessor.  I was wondering how you can get Grbl panel to run a macro whilst a program is running.  
It should be the grbl panel GUI that decides when to operate a tool change, the T# should trigger a tool change ?

MeJasonT, 2019-03-30:
Hi guys,
i have been trying to open GRBL-Panel in MS visual studio but it keep throwing up an error about repeatbutton.  I believe i need to reference it but i don't have a clue how to drive visual studio.  I'm also not sure what platform to use when building the code, i tried emailing Gerrit but he is no longer accepting emails via his github site.  Any pointers would be a great help.  I have tried reading up about adding references but the two project solution does not match the MS VS help examples.

Fusseldieb, 2019-04-09:
[CNCJS ](https://cnc.js.org/) is a nice alternative in my opinion, it let's you even define customized buttons which trigger user-defined G-Code.

1Metalguru, 2019-04-09:
GRBL Panel has those features as well, in the Macros tab.

andyb749, 2021-08-18:
> Sorry, this is not really a GRBL issue, but I didn't know where else to post it.
>
> I was just on the GRBL Panel Wiki page and it seems Gerrit Visser is hanging up his spurs and ending support for the [...]

This is something I'm definitely interested in picking up.... except I hate VB .Net and first change would be to port to C#.