gnea/grbl #535: GRBL V1.1G?

1Metalguru, 2018-10-07:
I have seen mention of a new version of GRBL, 1.1G, but I cant seem to find the download,

Anyone?

alk1913, 2018-10-07:
hello @1Metalguru, just download or clone the last update on this page, and follow the instructions in the wiki page to install it in the arduino IDE.
in the relase page the last version is 1.1f.

1Metalguru, 2018-10-07:
So the last update marked 1.1f is actually 1.1g?  I noticed there seemed to be 2 V1.1f's, is this just a type?
Why has nobody fixed this?

alk1913, 2018-10-07:
no, the last update 1.1g is not available in the release page.
so, yo need to download or clone it on this page.
no under elease page

1Metalguru, 2018-10-07:
I dont get it.  There are no downloads on this page...
I don't know what a clone is in this context...

alk1913, 2018-10-07:
[grbl-master.zip](https://github.com/gnea/grbl/files/2454366/grbl-master.zip)
this, and install follow thw instructions on wiki.

1Metalguru, 2018-10-07:
That's not compiled...  I have no idea how to compile it.  I am not a programmer.

Thanks anyway...

alk1913, 2018-10-07:
ok, in the wiki page, look for compiling grbl, and follow the instructions but using the zip that i send

chamnit, 2018-10-07:
Grbl v1.1g isn’t a release because it’s been in testing. When there is sufficient evidence that it’s stable, it’ll be posted in the release page.

closed 2018-10-07 by chamnit

AmroAsaad242, 2021-11-21:
is this control X Y Z axis and the servo ?