gnea/grbl #539: prescaler 256 in cpu_map.h

kolemjdouci, 2018-10-12:
Hello, would it be possible to add a line for 256 prescaler in cpu_map.h? I would like to use it to control my ESC.

 // Prescaled, 8-bit Fast PWM mode.
  #define SPINDLE_TCCRA_INIT_MASK   ((1<<WGM20) | (1<<WGM21))  // Configures fast PWM mode.
  // #define SPINDLE_TCCRB_INIT_MASK   (1<<CS20)               // Disable prescaler -> 62.5kHz
  // #define SPINDLE_TCCRB_INIT_MASK   (1<<CS21)               // 1/8 prescaler -> 7.8kHz (Used in v0.9)
  // #define SPINDLE_TCCRB_INIT_MASK   ((1<<CS21) | (1<<CS20)) // 1/32 prescaler -> 1.96kHz
  #define SPINDLE_TCCRB_INIT_MASK      (1<<CS22)               // 1/64 prescaler -> 0.98kHz (J-tech laser)

I'm new to arduino and I'm not sure how to do that.

Thank you

Kamil

RunnyBore604, 2018-10-12:
#define SPINDLE_TCCRB_INIT_MASK      ((1<<CS22) | (1<<CS21)) // 1/256 prescaler -> 244.140625Hz

kolemjdouci, 2018-10-12:
Thank you for your help

closed 2018-10-15 by kolemjdouci

kolemjdouci, 2025-12-09:
Hello, If I would like to set frequency 490hz, I should set 1/128 prescaler?

Like this:

#define SPINDLE_TCCRB_INIT_MASK ((1<<CS22) | (1<<CS20)) // 1/128 prescaler -> 488.28125Hz