gnea/grbl #543: Safety door priority?

bdurbrow, 2018-10-21:
I spotted this (I've posted code from the '328 version, it's a little different in the 2560 version but the issue is present there as well):

Because it's implemented with an "else if" tree, if any of the prior conditions are true, the latter ones aren't tested for.... and the safety door is checked for last.

Shouldn't it be checked for first, so that it takes priority over the other inputs, like cycle start?

From system.c, line 63:

ISR(CONTROL_INT_vect)
{
  uint8_t pin = system_control_get_state();
  if (pin) {
    if (bit_istrue(pin,CONTROL_PIN_INDEX_RESET)) {
      mc_reset();
    } else if (bit_istrue(pin,CONTROL_PIN_INDEX_CYCLE_START)) {
      bit_true(sys_rt_exec_state, EXEC_CYCLE_START);
    #ifndef ENABLE_SAFETY_DOOR_INPUT_PIN
      } else if (bit_istrue(pin,CONTROL_PIN_INDEX_FEED_HOLD)) {
        bit_true(sys_rt_exec_state, EXEC_FEED_HOLD);
    #else
      } else if (bit_istrue(pin,CONTROL_PIN_INDEX_SAFETY_DOOR)) {
        bit_true(sys_rt_exec_state, EXEC_SAFETY_DOOR);
    #endif
    }
  }
}

langwadt, 2018-10-21:
still looks like if cycle start and safety door happens and the same time, you'll get a  cycle start and ignore the safety door,  at least until the next pin change interrupt

chamnit, 2018-10-21:
Doh. These should be explicit if statements. I’ll correct it.

bdurbrow, 2018-11-16:
This needs to be applied to the grbl-mega repository as well.