gnea/grbl #546: GRBL crash/UART permanently sending data

Zorg3000, 2018-10-23:
I'm running a ArduinoNano where GRBL crashes after 2..20 minutes. It has nothing to do with the data I sent, so no matter what shapes I move and no matter what GUI I use, the problem is always reproducible. When the crash happens, I receive a lot of senseless data but GRBL does not react on anything except on an hard reset.

That's what I use

- no motors or anything else are connected to my Arduino, so there are no electrical interferences which could cause this

- USB is not used, I communicate with the ATMega directly via D1/TX and D0/RX pins

- serial communication is done with 57600 instead of 115200 (but changing the transmission rate to anything else does not have an influence on the problem)

- a 20 MHz crystal is used instead of a 16 MHz (F_CPU is of course set to 20000000 instead of 16000000 and in general UART-communication works properly - at least until GRBL dies)

I already tried different patches and modifications which completely disable EEPROM-accesses or change UART transmission from interrupt-based operation to polling mode, but none of these modifications had an influence on the problem.

I also tried different compilers (ArduinoIDE/Linux, avr-gcc/Linux, WINAVR) but this also did not change anything.

So my question: what could be the problem? Do I have to to anything more when running on 20 MHz beside of setting the F_CPU value? And how does it happen the problem does not occur reproducible but only after some (varying) time?

Any help and ideas are welcome!

chamnit, 2018-10-23:
Keep in mind that running at 20MHz is outside the supported build. It would help if you provided your Grbl $$ settings, $I build info, and any other config.h you have altered.

109JB, 2018-10-24:
How is the data sent? 
What is the streaming protocol used?
How often is the status query (?) sent?
What are your Grbl settings?

Zorg3000, 2018-10-24:
@chamnit I'm using the latest version from GIT with following build settings/parameters:

[OPT:HL,15,128]

$0=10
$1=75
$2=0
$3=0
$4=1
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=0
$24=30.000
$25=600.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=300.000
$101=300.000
$102=300.000
$110=600.000
$111=600.000
$112=600.000
$120=10.000
$121=10.000
$122=10.000
$130=5000.000
$131=5000.000
$132=5000.000

Zorg3000, 2018-10-24:
@cri-s thanks, I'll check the fuses! Unfortunately the board is running on quite a stable power supply, So I afraid this is not the reason :-o