gnea/grbl #549: Motors Not Moving

AcrimoniousMirth, 2018-10-30:
Hi,
When I send a move command from UGS, the motor powers on and locks so it can't be rotated by hand. It does not, however, move. I've tried this with new and old motors, two CNC shields and the DRV8825 drivers. I'm pretty sure step length is correct (10us), steps/mm is fine and acceleration I've also set low for testing this issue.
Any ideas?

X3msnake, 2018-10-30:
If you have access to a 5.usd logic analizer you can check that the driver pins are firing in the correct order to eliminate problems from the shield or the arduino.

[image]

I had a arduino clone that came with missing header pins and took us a long time to find the problem

[image]

chamnit, 2018-10-30:
@AcrimoniousMirth : Before you go and pull out a logic analyzer, please post your Grbl $$ settings, $I build info, and a brief description of your setup. It's probably much simpler that than.

@X3msnake : Thanks for the logic analyzer recommendation. Looks like it would be very handy. It looks much easier to see what's going on, over my oscilloscope.

X3msnake, 2018-10-30:
@chamnit Yeah it's a nice piece of equipment for the price. It was the only way we finally debbuged a DVD CNC we were trying to build that despite all settings being correct would not move.

The missing pin was the step pin.

@AcrimoniousMirth please do post the settings, @chamnit is right better to see if nothing is funky there first :)

AcrimoniousMirth, 2018-10-30:
@X3msnake @chamnit 

Thank you both! I kept playing around with it and I’m not entirely sure what I did but it started working. Probably just user error, the settings were default (as I’m just now getting back to retrying to build the CNC since stopping last year due to the same issue). Will look into getting one of those anyway, looks handy!

closed 2018-10-30 by AcrimoniousMirth