gnea/grbl #550: Failure to pull off of homing switch during homing

Blargedy, 2018-10-30:
I have the exact same problem as @rodfrey in [issue 265](https://github.com/gnea/grbl/issues/265), with the exception that I am using a [CNC xPro V3](https://openbuildspartstore.com/cnc-xpro-v3-controller-stepper-driver/) which has the stepper enable pins internal so rodfrey's solution doesn't work for me. 

Problem: When I run a homing command ($h), the moment the Z axis hits the homing switch an ALARM:8 is thrown. Sometimes an ALARM:9 is thrown, but almost always an ALARM:8

I have played with $24 and $25, but it doesn't even start to pull off of the homing switch. Regular jogs with G0 after an $X work fine. 

While stopped at the home position, a ? query shows the z axis limit pin being tripped.

Here are my configs:
$0=10
$1=50
$2=0
$3=1
$4=1
$5=0
$6=0
$10=1
$11=0.020
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=7
$24=1000.000
$25=10000.000
$26=50
$27=2.000
$30=1000
$31=0
$32=0
$100=106.478
$101=10.000
$102=10.000
$110=22050.000
$111=21600.000
$112=21600.000
$120=2000.000
$121=2000.000
$122=2000.000
$130=287.000
$131=720.000
$132=720.000

I flashed grbl_v1.1f.20170801.hex using XLoader, retrieved from the [releases page](https://github.com/gnea/grbl/releases). 

Thanks in advance for any help.

-Berni

chamnit, 2018-10-31:
There is no alarm codes 8 or 9. Just error codes. Error 8 is sending a command that requires an idle state but isn’t. Error 9 is a sending a command during an alarm state. 

I noticed your settings are very high. The x axis exceeds the 30kHz step rate limit that Grbl supports.

swarfer, 2018-10-31:
wow, I don't understand why X has such a high step/mm count compared to Y and Z.
and those are some seriously high speeds and accelerations, and homing at 1000mm/min is insane, it cannot stop in time to not overrun the switch by 'too far'.

so, I would turn off all homing stuff and get the calibration right for all axes first, then enable homing but much slower, like 1/10 of the current values.

swarfer, 2018-10-31:
and btw there is a calculator that will warn you about exceeding the max step rate...
http://swarfer.co.za/cnc/grblcalc.html

Blargedy, 2018-10-31:
thanks @chamnit and @swarfer for the replies.

@chamnit  
> There is no alarm codes 8 or 9. Just error codes. Error 8 is sending a command that requires an idle state but isn’t. Error 9 is a sending a command during an alarm state.

mmm. strange nomenclature then, to display ALARM: 8 if it is not alarm 8, but there is an alarm with error 8. 

> I noticed your settings are very high. The x axis exceeds the 30kHz step rate limit that Grbl supports.

I lowered the x axis speed to 4000 mm/min, which is a lot lower than I want it, but the positional precision is more important than speed. I still don't fully understand the step rate limit, but hopefully it is within tolerance now. 

@swarfer 
> wow, I don't understand why X has such a high step/mm count compared to Y and Z.

X uses a rack and pinion system and moves linearly, while Y and Z use a near 1:1 pulley system for rotational movements. The steps/mm have since been calibrated. However I can certainly see how they can look strange. 

> and those are some seriously high speeds and accelerations, and homing at 1000mm/min is insane, it cannot stop in time to not overrun the switch by 'too far'.
>
> so, I would turn off all homing stuff [...]

I lowered the homing speed since it was indeed excessive. It wasn't overshooting home by much if it was though. 

The issue was the placement of the homing switch. Since Y and Z are rotational, I am using a flipper that taps the switch from either side. The fact it had to be able to come from either side of the switch made the placement angle tricky.  I had the Z limit switch placed at the wrong angle and so the Z flipper had broken the limit switch as shown [here](https://imgur.com/a/HyYurhK). It was bent too far forward and so was not properly closing when the homing cycle was run. My high homing speeds and accelerations did not help this issue. 

I fixed it by lowering the speeds and accelerations, and replacing the z limit switch with a new switch placed more carefully. It seems to be homing reliably now. 

Thanks again @chamnit and @swarfer for the help.

Cheers.

closed 2018-10-31 by Blargedy

109JB, 2018-11-01:
> There is no alarm codes 8 or 9. Just error codes. Error 8 is sending a command that requires an idle state but isn’t. Error 9 is a sending a command during an alarm state.
> 
> I noticed your settings are very high. The x axis exceeds the 30kHz step rate limit that Grbl supports.

Not sure why you said this when the WIKI clearly states the following 2 alarm codes

8  Homing fail. Cycle failed to clear limit switch when pulling off. Try increasing pull-off setting or check wiring.
9 Homing fail. Could not find limit switch within search distance. Defined as 1.5 * max_travel on search and 5 * pulloff on locate phases.

chamnit, 2018-11-19:
@109JB : Not sure either. I must have been drinking that weird Kool-aid again.