gnea/grbl #552: Star Spindle automatically

mmsandrade, 2018-11-07:
Hi!
I´m using grpl 1.1 and Universal Gcode Sender 2.0.
I would like to start the spidle as soon as a send the file to Arduino. And the spindle stops when the work is done.
I have the hardware set up, my problem is in the sw. What should I do? I don´t want to do M4/M5....

109JB, 2018-11-07:
Since Grbl acts on streaming data, it has absolutely no idea when your job is starting or when it is ending. 

Why in the world would you not just want to add an M3 at the beginning of your file and a M5 at the end??

mmsandrade, 2018-11-07:
Of course grbl knows when the job started and when it is done. 
I have the project file, so he knows when to start and stop.

As soon the project file is downloaded to Arduino using Universal Gcode Sender the CNC start to work. The axis start to move, why not start the Spindle?

I don´t think it is smart to have a CNC machine that start moving without the spindle is on.

109JB, 2018-11-07:
Grbl on the Arduino doesn't get the entire file all at once. It only gets one line of a file at a time, so it has no way of knowing whether it is at the front of a job, the middle, or the end.

> I don´t think it is smart to have a CNC machine that start moving without the spindle is on.

I am sorry to be blunt, but I think it is rather stupid and more unsafe to assume that the spindle should be on whenever the machine is moving. Arbitrarily turning on the spindle is unsafe. 

There are many times that motion is required with the spindle off in CNC machines. For example when changing tools, or a programmed pause to check the part being machined, or a dry run, or a broaching operation, a scribing operation, a drawing operation, and many more.

It is a very very simple and industry accepted practice to use M3/M4 to start the spindle and M5 to stop it. In order to make every program you have start the spindle at the beginning and stop the spindle at the end requires adding only 2 lines of g-code to each file using a text editor. 

Bottom line is that what you are asking for is a very unsafe request.

swarfer, 2018-11-07:
GRBL follows the codes laid out at http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m3-m4-m5
which follows the Gcode standard which has codes
M3 for spindle on clockwise
M4 spindle on anticlockwise
M5 spindle off

further, M3 and M4 take  the S word with a speed in RPM.   GRBL uses this to set the PWM value on the spindle output pin.
for example
M3 S1000
could set the spindle on at 1000 RPM (if properly calibrated)

in other words, GRBL already supports the industry standard spindle control methods.
All YOU need to do is insert an M3 Sxxx at the start of your code and an M5 at the end AND connect your spindle to the PWM pin properly and you will have 'automatic spindle control'.

mmsandrade, 2018-11-08:
Thank you all for the explanation.

closed 2018-11-12 by chamnit