gnea/grbl #553: PWM output: 2ms cycle instead of 1ms

stawel, 2018-11-09:
Hi,
I'm currently trying to solve some issues when rastering images on my A3 pro engraver.
I've hooked up a logic analyzer to  stepper motor drivers inputs and PWM output, to see 
if the issues I'm experiencing are software or hardware related, and I wrote a very simple script to visualize the workflow,
if someone is interested:
https://github.com/stawel/grbl_output_analyzer/tree/master

I didn't solve my main problem yet  but I probably found a  bug in the grbl firmware:
It looks like the PWM output is occasionally overdriven (it's sometimes "on" longer than the 1ms when starting a new rastering section, sometimes it's 1.2ms, 1.9ms or 2ms long),
example (yellow lines represent PWM duty, see in full resolution):
[image]
On picture there are short bright yellow lines where they should't be.

here are screenshots from my logic analyzer:
[image]
[image]

gcode:
https://github.com/stawel/grbl_output_analyzer/blob/master/example/gcode.gcode

Best Regards,
Paweł

stawel, 2018-11-09:
my grbl configuration:
```
 $0=10
 $1=255
 $2=0
 $3=0
 $4=0
 $5=0
 $6=0
 $10=3
 $11=0.010
 $12=0.002
 $13=0
 $20=0
 $21=0
 $22=0
 $23=0
 $24=25.000
 $25=500.000
 $26=250
 $27=1.000
 $30=1000
 $31=0
 $32=1
 $100=80.000
 $101=80.000
 $102=80.000
 $110=10000.000
 $111=10000.000
 $112=5000.000
 $120=500.000
 $121=500.000
 $122=200.000
 $130=5000.000
 $131=5000.000
 $132=5000.000
```

stawel, 2018-11-10:
I've added a pull request which fixes this issue:
https://github.com/gnea/grbl/pull/554

The main problem was that the OC2A output (PWM output) was not cleared after the last rastered section, the new section occasionally stated with a high PWM duty even if it was dark.

with patch applied I'm getting this results (see in full resolution):
[image]

chamnit, 2018-11-10:
Thanks. I suspected it was the timer counter. There is an outstanding issue that is similar. I think your solution addresses both. I’ll check over it when I can. I kinda have my hands full at the moment.

stawel, 2018-11-11:
I did two more tests:
case 1: I only set TCNT2 to 0, without clearing OC2A
```
inline void disable_PWM()
{
     SPINDLE_TCCRB_REGISTER = 0;                       // Disable Timer (disable clock)
     SPINDLE_TCCRA_REGISTER = 0;                       // Disable PWM output.
     TCNT2 = 0;                                        // Reset Timer counter (ensures PWM synchronization)
}
```

case 2: I set TCNT2 to 0xff and clear OC2A (currently in pull request, previously TCNT2 was set to 0)

in both cases I get different results:
1.
[image]

2.
[image]

to my surprise case 1 is not synchronized along the X axis (every second line),
it probably has something to do with:
`All CPU write operations to the TCNT2 Register will block any compare match that occurs in the next
timer clock cycle, even when the timer is stopped.` (atmega328p datasheet page 193).

My conclusion is that case 2 is the best solution (or something similar), downside is that the synchronization will be a little bit delayed (will be of by 1/255 PWM cycle, probably negligible)

>  I’ll check over it when I can. I kinda have my hands full at the moment.

take your time, I'm glad that I can help,
Best Regards,
Paweł