gnea/grbl #555: Lathe / turning with grbl

yellobello, 2018-11-11:
Hi,

I have a small proxxon lathe (really small model building lathe) that I want to convert to cnc. (part-time)
My biggest concern is that I want to do some taping with it, and grbl doesn´t support it per se, since it has no encoder or motor position support. sad fact :-)
So, since the lathe´s motor is not very strong anyways I had the thought to swap it with an at least equally strong nema23/34 stepper or (hybrid) servo.
This way I could drive the spindle as an axis and do taping, dividing and more sun stuff with my tiny lathe.
The question is, is this supported by grbl and by any postprocessor or would I have to write my gcode by hand?

biasedlogic, 2018-11-12:
> I have a small proxxon lathe (really small model building lathe) that I
> want to convert to cnc. (part-time)
> My biggest concern is that I want to do some taping with it, and grbl
> doesn´t support it [...]
There are a few things to consider:
1. You will want to change the spindle drive to timing belts. AFAIR Proxxon
has a two-stage vee-belt drive, vee belts slip, so your threading would be
compromised by belt slip.
2. There's nothing wrong with keeping the timing belts for a day to day
use, as the small lathe motor will anyway stall when overloaded even with
vee belts, the drive could get a bit noisier with timing belts though.
3. as for CAM try Fusion 360, it has an integrated CAM module. I have never
explored what lathe options are there, but as it is Autodesk's full fledged
commercial CAM built in there, if that can't do what you're looking for,
you might be in for hand coding anyway.
4. Sure you know that, just for the others that may contemplate the mod:
Remember you will need tool retraction, as the leadscrew has play and the
lathe has some give to it, so you not only need to advance the tool in, but
fully retract it when getting back to start before another cutting pass. If
it wasn't for the play you could get away with 2-axis control (spindle and
longitudinal movement) and advance the tool by hand after each pass. So you
need to CNC the top slide (okay, for the small threads you can actually cut
on a proxxon you might get away with motorizing the cross slide only, but
you might get problems when trying to cut larger diameter threads)

-- 
Marek Lewandowski (thru google webmail)

109JB, 2018-11-12:
You may want to check out this thread in the grbl-Mega branch. user fschill has written code for spindle sync for grbl. Here is the github thread:
https://github.com/gnea/grbl-Mega/issues/26

Here is his code branch for the spindle sync;
https://github.com/fschill/grbl-Mega/tree/spindle_sync

And here is a video of it in action:
https://www.youtube.com/watch?v=Ia_g6kpPqtE&feature=youtu.be

langwadt, 2018-11-12:
when using a stepper or servo for the rotating axis threading is really just a line at an angle

bdurbrow, 2018-11-13:
This Old Tony just did a kinda similar thing, and has a video about it here:
https://youtu.be/mAZei3wpBe0

He's not using grbl, and it's an add-on to his CNC router, but the idea is the same.

As for CAM, I think you can generate a post-processor for Fusion 360 that will do that? I havn't dug deeply into Fusion's post processor engine, but IIRC, it's based on Javascript; and is quite flexible.