gnea/grbl #557: Adding Tool Table Parameters

bdurbrow, 2018-11-13:
I'm about to add a tool data table (length offset & radius) to the work I've been doing on the grbl-mega fork... (This data would be stored in eeprom, along with the other settings.)

I'm thinking of making the tool data available via the $ parameters, starting at $500 for tool 0 length, $501 for tool 1 length, etc; and $600 for tool 0 radius; $601 for tool 1, etc.

I'm going to use the data for probe offsetting; but in the future I can see these parameter assignments being used for tool radius compensation.

I don't think that there's enough flash space for the code on atmega328 based platforms, but there's plenty of room on atmega2560 and the upcoming SAMD based platforms... and more than enough eeprom space for a really large tool table, and still have plenty of room for expansion.

Any comments? Idea? Rotten tomatoes?

chamnit, 2018-11-13:
@bdurbrow : I had thought about adding tool data in the past. Having tool length is useful, but I found that it was difficult to come up with something universal while accommodating the variety of different tools and user workflows. In addition, storing tool radius won't help you with anything since Grbl doesn't have cutter compensation. 

So instead, I designed Grbl to accept the dynamic tool length compensation command G43.1 only. This puts the tool data library in the hands of the GUI, which can be customized much easier than altering Grbl's firmware. The GUI has a bit more responsibility in managing the tool data, but it can also do other things like perform the commands for a tool change and offset probe via a scripted set of code commands streamed to Grbl. This gave the flexibility that I was looking for. While not completely ideal, it seems to work.

bdurbrow, 2018-11-14:
Doing it all in the GUI works great... if you have a GUI. I don't - I'm streaming the g-code directly from the SD card reader attached to the RAMPS card. The whole idea is to not have a PC attached to the CNC machine (the other side effect is that manual operations - like jogging - are a *lot* more responsive).

I use the radius data in the setup where you have the cutting tool in the spindle while probing (typical of a CNC router, as opposed to an electronic touch probe tool used in more conventional setups, like on a VMC). Then, the touch plate thickness and tool radius is taken into account when setting G54/G55 etc...

Anyway, when radius compensation *is* added to grbl, does the interface to the tool data I described make sense to you?

chamnit, 2018-11-14:
Your RAMPS card is a UI. It can do the same job managing a tool table as a full GUI. 

Tool radius comp is more complicated than it looks. You need the data of the entire gcode program to correctly calculate the offsets and edge conditions. Grbl can’t do that since it receives a stream of the program. Cutter comp is something a UI will also need to be responsible for.

shooter64738, 2018-11-16:
I didnt implement crc natively in grbl. I wanted too, but it was just going to be too difficult. The look ahead was my primary obstacle. So I have implimented it on a second arduino that compensates the motion based on a tool table. I have a detailed write up on how it works, and why look ahead is the linch pin for it. Also as  chamnit said, its not quite as simple as it seems. I dont read all of the gcode, but it must read ahead atleast 1 MOTION line. The more, the better path crossing detection will be.
https://github.com/shooter64738/CRC/wiki
Its been 2 years to get it this far and the code is still getting tweaked.

shooter64738, 2018-11-16:
There are a few places where the methods have changed, and they don't follow the wiki to the letter now, but for the most part it does. I tried to find a way to do it without looking ahead, but it became impossible, so at least one motion line must be read within 10 lines of my buffer. G0/1/2/3 could be the first motion, and you could have 8 additional lines following for G or M codes that don't cause motion, but the 9th line has to cause a motion, or the look ahead will fail. I really dont know of any other way you can do it reliably without at least 1 look ahead motion, but I am by no means an expert on the subject.

shooter64738, 2018-11-16:
> As the radius compensation have a final close motion, the motion can be shorted by 1.2 times the radius compensation and the unfinished motion added to the next or final closing motion. 2018-11-16 6:42 GMT+01:00, shooter64738 <notifications@github.com>:

That is essentially what look ahead is doing. It is using point data of the forward motion to determine where the current motion should end. You could start motion on the current path, but I could never come up with a way to determine the current path ending without knowing what the next motion was going to be. Starting the motion, stopping short, and waiting for the next motion on an arc winding CW that connected to another arc winding CCW at an outside corner was a mathematical nightmare. It became easier to leave the machine at rest until the next motion came in and then start and end the current motion. In either case, I still needed the forward motion data to determine where to end the current motion.