gnea/grbl #558: Parsing incoming messages from GRBL

MeJasonT, 2018-11-13:
I was wondering if some kind person could write a basic parse routine for me to run on another Arduino which would sniff the data from the PC/Grbl communication Link,  By using a second arduino i can receive the return messages from the grbl TX pin - This is not affecting the operation of the pc/grbl control.
Im currently able to read the return string and output it via the arduino IDE serial monitor.

I would like to be able to parse the data into an array (block) so i can manipulate the values returned for Spindle speed and the operation of suds/coolant pump and the spindle direction.  I am attempting to make an interface between grbl and my milling machine motors variable frequency drive.  To cut a long story short - the high frequency noise from the VFD upsets the grbl operation, im wanting to transmit the commands over RS485 between the board sniffing the signals and the VFD.  The return message has the beauty of telling me when a function has been performed and not when its being queued which is what i would get from the other side of the comms link. 
Being able to do this would also benefit those wishing to build a torch height controller for a plasma table or any other project where reporting would be an advantage as its not possible to provide any further resources from grbl to transmit from a second serial port due to the maxed out memory especially on the UNO variant.

 I can see its possible and have the ability to do everything apart from parsing the message string,  any help would be most appreciated.

chamnit, 2018-11-13:
@MeJasonT : Since this isn't a direct Grbl issue, I'm closing this thread, but people can still respond to your request.

closed 2018-11-13 by chamnit

MeJasonT, 2018-11-13:
Thanks Sonny

MeJasonT, 2018-11-13:
On a side note - When are us clankys going to get a point and click version of GRBL for M0, its still cloak and dagger amongst the programming fraternity.

MeJasonT, 2018-11-14:
For anyone interested.
uses Nick Gammon's RS485 sketch and library

`//====================================================================
// parseGRBLRS485.ino
// Original GRBL Parse code with RS485 comms added
// Parse code for GRBL CNC
// 13/11/2018
// Ver 1.0
//====================================================================

#include "RS485_protocol.h"
#include <SoftwareSerial.h>

//  RS485 Communication variables /////////////////////////////////////

SoftwareSerial rs485 (2, 3);  // receive pin, transmit pin
 
const byte ENABLE_PIN = 4;
int channel = 0;
int deviceID = 0;
int function = 2;
int data = 0;

const byte LED_PIN = 13;

// Parsing GRBL message variables  ///////////////////////////////////////////

const byte numChars =71;
char receivedChars[numChars];
char tempChars[numChars];        // temporary array for use when parsing

//Feed and Speed variables
char FSMsgGRBL[12];
int FS_FeedMsg = 0;
int FS_SpeedMsg = 0;

//Overide variables
bool OvMsg = false; //Ov message being transmitted
int CmdCoolant = 0;
int CmdSpindleCW = 0;

// Variables to hold Message blocks

char GRBL_msg1 [numChars] = {0};
char GRBL_msg2 [numChars] = {0};
char GRBL_msg3 [numChars] = {0};
char GRBL_msg4 [numChars] = {0};
char GRBL_msg5 [numChars] = {0};
char GRBL_msg6 [numChars] = {0};

boolean newData = false;

//============================================================================

// callback routines RS485 Comms
  
void fWrite (const byte what)

      {
      rs485.write (what);  
      }
      
    int fAvailable ()
      {
      return rs485.available ();  
      }
    
    int fRead ()
      {
[... 233 lines cut ...]
//                     //RXFunc,
//                     //RXdata1,
//                    // RXdata2,
//                      
//               };
               
              //delay (10);  // give the master a moment to prepare to receive 
              ////Serial.println(buf2[2]);   
              //digitalWrite (LED_PIN, received == 0);  // turn on LED if error
              //
              //        if (buf [1] >= 20){
              //        
              //               InputDevice= buf2 [1];
              //               Inputvalue = buf2 [2];// write coil value to a variable
              //               Serial.print ("A"); Serial.print (InputDevice); Serial.print (","); Serial.print (Inputvalue);Serial.println ("!");
              //        }
              //
              //   // old_level = data;
          
          
         // }  
  
  }// End of DoBusComs
  
`

MeJasonT, 2018-11-14:
I have no idea why it displayed the code like this, but its all there - unlike me

MeJasonT, 2018-11-19:
Follow link here
[https://forum.arduino.cc/index.php?topic=578683.0](url)

jschoch, 2018-12-07:
I used some existing work and put it in an arduino library

It puts the data into a handy struct.  You just need to pass it the string from serial.

https://github.com/jschoch/grbl_chat