gnea/grbl #561: D11 is always 0V

muxa, 2018-11-17:
When i run `S255 M3` i get 0V output on D11. Here are the snippets of my config:
```
$30 = 255    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 1    (Laser-mode enable, boolean)
```
Running grbl_v1.1f.20170801 in Arduino Nano. I have checked D11 with a test sketch and it works. 

Any ideas?

swarfer, 2018-11-17:
yeah, read the Wiki!
laser mode https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode
in laser mode the PWM is enabled only  during movement

muxa, 2018-11-17:
Thanks, i did not realise that G1 can be specified without coordinates. 

The following worked:
```
G1 F10
```
Turn on laser for focusing:
```
S1 G3
```
Turn laser off:
```
G5
```