gnea/grbl #562: error 34

gl40, 2018-11-18:
Hi,

Trying to execute attached code, I get this error 34 : 
"Dernière erreur : g3x28.7487y61.4804r2.4643

Un arc G2 ou G3, tracé avec le rayon défini, retourne une erreur mathématique lors du calcul de la géométrie de l'arc. Essayez soit de casser l'arc en demi-cercles ou quart de cercles, ou redéfinissez-les en définissant le décalage d'arc."

The code was generated using (the excellent) bCNC and seems OK to me.
I tried different card (uno) and grbl version (1.1e, 1.1g) with same issue.

No idea with I could do next 

Thanks for any help with that.

[puzzle.nc.txt](https://github.com/gnea/grbl/files/2592868/puzzle.nc.txt)

zeevy, 2018-11-19:
```
A G2 or G3 arc, traced with the radius definition, had a mathematical error when computing the arc geometry. Try either breaking up the arc into semi-circles or quadrants, or redefine them with the arc offset definition.
```

you should avoid any G2 or G3 arcs if those are not compatible with grbl

bdurbrow, 2018-11-19:
My guess is that it's due to the R form of the arc being ambiguous. Try the IJ form instead.

MeJasonT, 2018-11-19:
What program/post processor are you using to create the .nc code.
you need a post processor compatible with GRBL.  If im telling you how to suck eggs, i apologise but in case you don't know a post processor is a cod script which converts your drawings coordinates into commands compatible with a cnc machine.  Eg, i use Cambam to convert my cad drawings into machining processes, cambam has a GRBL post-processor so it converts any tool paths it creates into NC  code compatible with GRBL.  Your puzzel code is not recognised by GRBL as the NC code is incompatible.  You may have had code work before and this is the first time you have used arcs so GRBL has spat out the dummy.  What is your tool chain ie how do you get to produce your NC file, what cad program do you use, how do you produce CAM.   Hopefully we can point you in the right direction with an idea of the programs you use. OK I read bCNC above, i'm not familiar with it so i will run it and check your example as to why it isn't outputting in GRBL compatible code.  I use a freeware program called camotics to check and visualise my code, unfortunately it wouldn't have picked this one up as it is a valid NC code. (just not for GRBL)

MeJasonT, 2018-11-19:
Is your hardware selected as smoothie or grbl in bcnc settings.
I cant get python to run bcnc yet, not another bloody language to learn - i'm one of the spoilt generation who likes point and click, whats this writing script in command boxes to make stuff work.  Reminds me of good old DOS 3.1 days.

chamnit, 2018-11-19:
R radius mode G2/3 arcs are notoriously bad. They pose problems for all controllers because the way the radius problem is mathematically posed. It’s extremely sensitive to small number errors and round off. Always use IJK mode instead. 

Grbl follows the LinuxCNC standard for error checking arcs. It’s actually pretty lax. Just be glad that this check is there. Otherwise you’ll have parts that are off by several thousands or more than a tenth of a millimeter.

closed 2018-11-19 by chamnit