gnea/grbl #563: Intermittent failure to home. Alarm: 8.

Blargedy, 2018-11-19:
Hello,

I have previously had [this issue](https://github.com/gnea/grbl/issues/550) but I am now thinking the old thread solved a different problem which was also giving trouble. The system is responding in the same way as before. During a homing cycle, the moment the z-axis limit switch is hit the homing cycle fails with ALARM:8 being shown on the terminal. 

Sometimes moving the axis a bit away from the limit switches and retrying the homing cycle works, but many times it just refuses over and over. 

Here are my settings: 
$0=10
$1=50
$2=0
$3=1
$4=1
$5=0
$6=0
$10=1
$11=0.020
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=7
$24=50.000
$25=500.000
$26=50
$27=2.000
$30=1000
$31=0
$32=0
$100=106.478
$101=18.300
$102=18.300
$110=4000.000
$111=2160.000
$112=2160.000
$120=10.000
$121=20.000
$122=20.000
$130=2875.000
$131=720.000
$132=720.000

Thanks in advance for any help.

-Berni

chamnit, 2018-11-19:
Sorry about the last issue. My mistake that there isn't an alarm 8 or 9. I think I was looking at a stripped down side project that day, where I removed a lot of functions. 

You probably need to increase the $27 pull off setting. Alarm triggers after the homing cycle finds a switch and tries to move off of it to disengage it. If it detects the switch at the end of the pull-off motion, it'll invoke the alarm 8. 

If you are physically off the switch at the end of the pull-off motion and still get this error, that means Grbl is still detecting a signal on the limit input. This is often caused by electromagnetic interference, bad wiring, or a bad ground. It's a common issue when dealing with switches.

Blargedy, 2018-11-20:
I had to increase $27 to 15 to get a reliable home. I had originally increased it to 8 and figured that that should be enough, so when it still failed I ruled out the homing pull-off. Thanks for the tip. 

-Berni

closed 2018-11-20 by Blargedy