gnea/grbl #566: Stair Stepping XY on diagonal lines.

timryder, 2018-11-28:
I am getting some very fine stair stepping when I do a diagonal line between my X and Y axes.  
I am using the Arduino Uno with 0.9 build of grbl. (I know it's old) do you guys know what parameter if any can be adjusted to help with that? the mechanics are dead on I have verified their slop and repeat ability to be even tighter than the stair stepping I am getting so pretend for now that that's not the issue.  Any value or config I can check/change to improve that? 

Apologies if this is the wrong place to post the question, I didn't really see an open forum... I'l\l delete it when I resolve the issue.
Thanks!

X3msnake, 2018-11-28:
microstepping on your stepper drivers.

check a example of this about halfway of the video

https://youtu.be/ZzO9skEYqwY

timryder, 2018-11-28:
Didn't seem to make a difference, I am using 1/16 by default and switched to 1/32 to see if it would eliminate it, but it did not.  The exact same stair stepping pattern is present in the linear move. not sure what's causing it.  I was wondering if there was some hidden firmware trick or some setting in the config.  I have my Stepper pulse width at 10us. I'm curious if the mcu technically pulses the individual axis one at a time just super quickly or are they pulsed at the exact same time.

chamnit, 2018-11-28:
@timryder : Grbl pulses multi-axis steps at the same time based on the fastest axis, but slower axes are delayed precisely by the bresenham line algorithm. Altering the step/mm of your axes will improve the accuracy. You are probably seeing something else, like dirty timing belts. You'll have to show a picture of what you are calling stair stepping to get more help.

X3msnake, 2018-11-28:
What are you using as CAM ?

Your steps can be derived from the cam software precision also. Just drop
your gcode on this online tool and see if the steps show there

https://ncviewer.com/

à(s) 20:07: