gnea/grbl #567: simplify coding

cjren92, 2018-11-30:
Hello,
This is my first project involving any coding or using arduino. i have my motors working and tuned to move 1mm(y) and turn 1Degree(x)

Y-comes back and x-rotates before Y goes back in, This process is then repeated adding 10 degrees to X. 
G01Y-20F300
Y3
y-20
x10
y3
y-20
x20
y3
y-20
x30
y3
y-20

My code has many lines and i am typing these by hand, is there a better way to do this?

carneeki, 2018-11-30:
Many of us use software to stream GCode to GRBL. Personally I'm a fan of
Universal G-Code Sender, but others use bCNC and Chilli. The small subset
of G-Code that GRBL supports doesn't include "canned cycles" which are
routines to make small reusable pieces of G-Code. GRBL philosophy is driven
a bit by the constraints of the hardware in that GRBL should support basic
motions, and more complex things should be handled by software that
generates the more complex code (mostly we just point and click at holes in
some CAD / CAM software and it spits out many lines of XY coords and then
does drilling on the Z axis for us). Most GRBL users tend to run machinery
that plots / prints / cuts metal on 3 axes, your application sounds a
little different (perhaps a wire bender / robotics / some sort of
repetitive machine?).

One option you could maybe use is to change from absolute distance mode
(G90) to incremental (G91), I notice in the example, you pull back Y-20
then push it to Y3 once before starting the repetitive block. If there's an
important reason for this, then you would need to separate the code into
two parts, the "one shot starter" file that is streamed once, and then a
second file that you can stream repeatedly.

G91 G01 Y-20 F300 (do this once, G91 sets incremental)
Y23 (the one time push forward without rotate)

Y-20 (these 3 lines can be repeated)
X10
Y23

If it was just a typo and there's meant to be a rotate on line 2 you could
put it all in a single file:

G91 G01 Y-20 F300
X10
Y23

Then just press play on a G-code streamer, or if you're feeling
adventurous, a Python script to stream this to GRBL over and over (this
might requite some help from a Python forum), but GRBL has two sample
scripts to get you started (I haven't used them):
https://github.com/gnea/grbl/tree/master/doc/script
There's stream.py and simple_stream.py.

If you need to have a script that does the one shot start, and then repeat
you'd basically need to do something like
1) stream the starter file.
2) start a loop (set some sort of end condition, like "run 100 times" or
"run forever").
2a) stream the repeated file in the loop

Both approaches assume that you've got some sort of way to set the machine
up in it's starting position and also set the coordinates (this is called a
homing process).

Best of luck!

cjren92, 2018-11-30:
Thank you Adam!
The reason for pulling back to y-20 was that i start with my part at 0 against the tool. This seemed like the easiest way to easily replicate 0. The Y+3 should make a 3mm Cut. The first section doesnt rotate because there are no cuts yet. I can take that section out to make things easier, just repeat one more time.
The incremental setting should make things much simpler to code, even typing by hand :D
Thanks again.