gnea/grbl #569: GRBL crash, suddenly stops moving

beckmx, 2018-12-01:
Hello, I noticed that the GRBL version 1.1g from the Arduino library:
// Grbl versioning system
#define GRBL_VERSION "1.1g"
#define GRBL_VERSION_BUILD "20180614"

stops responding after around 60 commands sent, the file I use is this one
[part2.txt](https://github.com/gnea/grbl/files/2635725/part2.txt)

Should I upgrade or downgrade?

PicEngraver, 2018-12-01:
Works for me in Grbl 1.1g and in Mach2.  Is it Imperial or Metric?  I assume metric, but your file has no G21. 
Do you get any error messages?

PicEngraver, 2018-12-01:
File runs here in GRBL 1.1g and Mach3.  Check your sender program and Windows(?) USB power settings.  Windows may be putting USB port into sleep mode.

chamnit, 2018-12-01:
Your gcode is full of extremely short line motions and lots of redundant positions in the commands. Grbl can handle about 250 lines per second. Because your commands are so short (and shorter than your machine precision), you are likely exceeding this limit. It would also help if you post your $$ Grbl settings to see if your running Grbl too fast.

beckmx, 2018-12-01:
Ok, I left the probe connected, looks like the interaction with the spindle makes the Arduino crashes, after disconnecting everything from the PCB it ran great

closed 2018-12-01 by beckmx