gnea/grbl #572: Homing cycle.

sfinexer, 2018-12-05:
have $27.
Also
#define HOMING_CYCLE_0 ((1 << X_AXIS) | (1 << Y_AXIS))
#define HOMING_CYCLE_1 (1 << Z_AXIS)
Is it possible to make the search different?
Initially, the search for X, Y is further search for Z but so that between these cycles there is no Homing pull-off.
Ie reaching X, Y, then go to Z?

langwadt, 2018-12-05:
makes no sense to not home Z first, homing X and Y without knowing where Z is is a recipe for crashing

sfinexer, 2018-12-05:
@langwadt 
You contradict:
#define HOMING_CYCLE_0 ((1 << X_AXIS) | (1 << Y_AXIS)|(1 << Z_AXIS))

langwadt, 2018-12-05:
it does not make sense to not have #define HOMING_CYCLE_0 (1 << Z_AXIS)
if Z is not at the top of it's travel when you home X and Y you risk crashing into things