gnea/grbl #573: Spinde (laser) did not turn off when paused (hold)

sinfocomp, 2018-12-06:
Put machine on hold by input switch to hold job becouse laser head crashed and laser did not turn off, i burnt my shirt !!!!!

carneeki, 2018-12-06:
Feed hold is *not* an emergency stop. The documentation states:

> By machine control definition, a feed hold does not disable the spindle or coolant. Only motion.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands

I would need to check the source code to confirm, but I take it to mean that spindle control and laser power are synonymous for the purpose of machine state.

This same page also mentions a safety door pin (and real time command to trigger via software) which exhibits similar behaviour to feed hold in that the motion will hold, the spindle will shut off and coolant will also stop. This is still not an E-stop, but it might be what you're after as the machine won't lose its state like a sudden power off or reset. I've not used the safety door feature, but the documentation says that when the door is closed, GRBL will return to an IDLE state which means you can press CS / resume and the machine will go back to where it was, spin up and continue moving along.

Carl-jackslabs, 2019-03-19:
Great for pointing this out - very important to know

Perhaps this is a situation to wire a manual spindle speed switch (Could use a DPDT E-stop along with the hold signal line) for this kind of situation? I've always used a mill where I can manually and physically switch the tool head signal on/off. Would this work?