gnea/grbl #574: error 33 with large I or J values (?)

Cdriko, 2018-12-09:
hello
I use gcodetools derivative in inkscape to generate gocodes.
Sudently, using dot lines, it generates a lot of G3 paths with very large radius instead of just lines.
The result is like that :
`;(...)
G0 X33.9176 Y243.2292

M03
G1  F3500.000000 S255
G03 X36.2921 Y243.2228 I237.1478 J52084.2875 F3500.000000 S255
G1  F3500.000000 S255
G03 X39.0374 Y243.2154 I-24.9066 J98219.9788
M05

G0 X39.6771 Y243.2126

M03
G1  F3500.000000 S255
G1  X44.7973 Y243.1988
M05

G0 X45.4372 Y243.1959

M03
G1  F3500.000000 S255
G02 X48.1825 Y243.1886 I2604.9744 J-485351.2826
G1  F3500.000000 S255
G03 X50.5575 Y243.1824 I41.402 J17696.8596
M05
;(...)`

But at each G02 or G03, I've got an error 33.
Maybe it's because the large values for I and J ???

how fixing that ?

Cdriko, 2018-12-10:
thanks. 
So I've made a python loop to truncate the long floats. at the begining, ti seems to work but at the line 1626, I've the same error after this line :
`G1  F3500.000000 S255
G02 X56.6168 Y325.5949  I-97892. J-102606
G1  F3500.000000 S255
G02 X56.443 Y324.8939  I-49900. J64662.5`

Maybe even the than values are shorten than 7 numbers, the result of the interpolation overflow a stack ?

swarfer, 2018-12-11:
you should rather convert those G02 to G01 like this
G1 F3500.000000 S255
G01 X56.6168 Y325.5949
G01 X56.443 Y324.8939

notice that the actual distance moved is very short, 0.72mm and you are not going to see the difference between an arc and a straight line.

even better is to tell your Gcode generator to do this for you, ie, generate lines (G01) for any segment shorter than 1mm and radius larger than, say,  5000mm