gnea/grbl #576: 5 lines of gcode reproducibly crashes GRBL 1.1

igi79, 2018-12-16:
Hi, following 5 lines of g-code (part of a much larger file) reproducibly crashes GRBL 1.1f (grbl_v1.1f.20170801.hex) :
`S5000 M3
G0 X61.569 Y16.0
G1 X41.583 Y18.051 F170
G1 X41.646 Y18.058
G1 X41.672 Y18.059
G1 X44.451
G1 Z2.0
G0 X28.358 Y3.559
M5
`
with following settings:
`$0=10
$1=255
$2=0
$3=1
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=6400.000
$101=6400.000
$102=6400.000
$110=375.000
$111=375.000
$112=375.000
$120=15.000
$121=15.000
$122=15.000
$130=200.000
$131=200.000
$132=200.000`

Tested on latest and stable versions of bCNC and UGS. GRBL answers with some interesting >127 ascii characters and at the end is:
`787|FS:0.0,0|Pn:P|Ov:100,100,100|A:S`

After spending a day making a scrap material, I flashed a 'fresh new' bare atmega328p (with just 16MHz crystal) with latest HEX and connected with FTDI cable, the same error showed again!

Later I found, that lowering steps/mm down to 6000 helps.... I'm using DRV8825 with 1/32-step, which runs motor really smoothly so going down to 1/16 (and 3200 steps/mm) really unfortunate.

I don't think, this is the limit of microcontroller - issue is there  even when this g-code is run on 25% feed rate! On some feed rates like 25% - the crash behaves quite dangerously - X-axis keeps running!

langwadt, 2018-12-16:
6400*375/60 = 40kHz GRBL is only rated for 30KHz

igi79, 2018-12-16:
Thank you.
So maxrate = 30000*60/stepspm,
which for 6400 steps/m is 281mm/m and for 3200 steps/m 562mm/m.