gnea/grbl #577: Does grbl work on Ardunino uno wifi

thawkins, 2018-12-18:
Before I go off and buy one, does anybody know if grbl will work on the Arduino Uno Wifi with a grbl shield?. 
It appears to be an uno clone with the following differences.
1. Wifi hardware integrated (obviously), hardware tcpip stack
2. 48k vs 32k flash
3. 6k vs 2k sram
4. 256 bytes vs 1k eeprom
it has a few other things probaly not relevant to grbl, like an IMU (inertial management unit, whatever that is), and crypto processor, also used by the wifi function. 
Its based on a new Microchip clone of the 328, with more flash and sram, although i was not sure tbe reduced size eeprom woukd be an issue. It would make a good half way house between a 328 and a 2560, more space for code without all the pin remapping.

bdurbrow, 2018-12-25:
If you want WiFi, I would suggest a Mega2560 and either an ESP8266 or ESP32 board. I would also suggest considering a RAMPS 1.4 + character LCD instead of a standard grbl shield; I've got a fork of the Mega2560 version of grbl that I'm working on that supports streaming from the SD slot on the LCD board as well as some other features; extending that to support WiFi via esp card won't be difficult.

PCSDIAS, 2019-01-28:
If you want wireless communication you could try a HC05/HC06 Bluetooth module connected directly to Arduino Uno TX and RX pins.
Paulo

HuubBuis, 2019-01-28:
I use an arduino uno ESP8266 module for the GRBL controller.  It works more reliable than the  Bluetooth solution. I started using a separate ESP8266 module but now the are Arduino Uno and Mega versions having a buildin ESP8266 board [https://robotdyn.com/uno-wifi-r3-atmega328p-esp8266-32mb-flash-usb-ttl-ch340g-micro-usb.html](url) . I have the Uno version of this board and ordered the mega version.
I have written a sketch to run the ESP8266 as transparant WiFi bridge and a manual for programming the ESP8266 just look here: [http://www.mwt.messageboard.nl/forum/viewtopic.php?f=4&t=11](url)

PCSDIAS, 2019-01-28:
@HuubBuis Nice solution.
What controller software do you use in order to stream the g-code to the GRBL via WiFi?

HuubBuis, 2019-01-28:
I have tested it using my own software and Grbl-Panel [https://github.com/Gerritv/Grbl-Panel/wiki](url).

If you add a separate ESP module on the PC's USB port that connects (automatic) to the ESP module on the Arduino, you can use all GRBL GCode senders even if they don't support IP communication.

PCSDIAS, 2019-01-28:
@HuubBuis A custom solution as I had imagined. I used the Bluetooth module because I wanted to be able to use any control software available for the GRBL, with a smaller part counrnumber of components.

Oldie-driver, 2019-01-29:
For wifi and streaming, i use a raspberry pi 3+ with cncjs. So i have a webserver with cam and can send from desktop the cnc files to the raspberry. 
The little miller is 50 meters away in a garage, connect over powerline adapter and wifi for the last meter.
For preparing the work, the raspberry got a cheap touchscreen for jogging / homing. 

The Arduino is connect to raspberry over USB. Streaming just in time over 50 meter powerline and then wifi may to danger, i fear missing / corrupt data.

Just check on github for cncjs. Is a bigger solution then just a wifi module, but less work then use a Atom mainboard with vnc.

Quajetson, 2019-02-07:
If u want wifi look at this, wifi version is available
https://github.com/makerbase-mks/MKS-TFT
support grbl and custom commands/ buttons

Quajetson, 2019-02-07:
Tested on woodpecker board:)

manojprasanga, 2019-07-26:
can we use  Arduino UNO R3 D1 R32 microcontroller to control cnc v3 shield
[image]
[image]

Quajetson, 2019-07-26:
No. This is ESP based device.

manojprasanga, 2019-07-26:
But some one says most Arduino shields compatible with Arduino UNO R3 D1 R32 microcontroller.
Refference:-https://www.instructables.com/id/The-Greatest-Arduino-UNO-in-the-World/

luc-github, 2019-07-26:
@manojprasanga you can check https://github.com/bdring/Grbl_Esp32 but I do not know if such config is supported

manojprasanga, 2019-07-26:
@luc-github thanx

atrejo, 2021-03-05:
the answer is yes, the board and the wemos d1r32 and the cnc shield work at least with the a4988 drivers, the problem is with grbl https://github.com/bdring/Grbl_Esp32 I cant make it work with the wemos, but I think this sw problem is easy to workaround