gnea/grbl #578: GRBL for ATXMega?

Zorg3000, 2018-12-18:
This is more a feature request than an issue:

On a standard Arduino with 16 MHz clock GRBL provides a maximum step frequency of about 30 kHz, when one replaces the Oscillator by a 20 MHz variant, the limit is somewhere at 37 kHz. Meanwhile this is beyond the lower limit also for hobbyists and DIY-projects.

One easy solution which would come into my mind: use an ATXMega with 40 MHz clock would boost the maximum step frequency to about 75 kHz which is far beyond other CNC controllers but an easy to do improvement.

I already tried to compile GRBL for an ATXMega, but it seems all register names are different there, so some more work would have to be done which is beyond my knowledge.

So perhaps anybody can implement ATXMega support simply by replacing the register calls/names and encapsulate everything in some #ifdefs, so that it still can be compiled for "normal" ATMegas too? This would be a quick and _not_ dirty solution which boosts the performance of GRBL.

chamnit, 2018-12-18:
XMega would definitely add some boost and space to grow, but it isn't exactly an easy to find chip by DIYers. XMega is also way slower than current ARM M0/3/4 chips,  which are far more accessible. So there isn't any reason to invest a lot of time and effort into XMega.

closed 2018-12-18 by chamnit

Zorg3000, 2018-12-19:
OK but the existance of an ARM-GRBL is theoretical and the hitch to port GRBL to ARM is much higher - so the ATXMega port would be something which could be done in finite time!?

Zorg3000, 2018-12-20:
Where can one find this LPC1769 port? Only thing I have seen are some issues here in this bugtracker from beginning of this year where it was stated ARM port is nearly finished - but that's all...

mayhem2408, 2018-12-20:
https://github.com/gnea/grbl-LPC. I run this on a Smoothieboard and it works really well. Very fast.