gnea/grbl #579: GCode error after resuming pause

Salamandar, 2018-12-18:
I'm using GRBL1.1f with CNCjs. 
I keep getting "invalid GCode ID" errors and the CNC moves are missing some GCode lines. 
After having investigated electrical interferences or serial issues ( https://github.com/gnea/grbl/wiki/Known-Issues ), and replacing the arduino, using a raspberry pi,… we concluded the problem was appearing after pausing/resuming the job. (some lines after the resume, probably due to the command buffer)
What could cause those errors ?

chamnit, 2018-12-18:
It would help if you shared the error ID codes and the g-code lines that are problematic. Usually these are associated with unsupported g-codes or invalid G2 and G3 arcs output by CAM programs (older Fusion360 files in particular).

Salamandar, 2018-12-18:
Will do tomorrow. I will reproduce the bug after flashing the arduino with a debug-instrumented firmware.
I'm using Freecad and the same gcode line is usually interpreted correctly if no pause/resume is requested.