gnea/grbl #580: After upgrading, measurements are off.

TheRealVaranox, 2018-12-18:
I upgraded to GRBL 1.1f, and after doing that, when I tell the laser engraver to move 50mm on the x axis, it moves about 150mm.. Any Ideas?

109JB, 2018-12-18:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

TheRealVaranox, 2018-12-18:
So, that doesn't help. like at all... How do I access the firmware to change values?

TheRealVaranox, 2018-12-18:
Thanks for nothing ya pricks

chamnit, 2018-12-18:
It says at the top of both wiki pages. Connect to Grbl via a serial terminal program. It walks you through   the interface and how to change things. Please try that first before passing on blame.

109JB, 2018-12-19:
So you don't know what you are doing and can't be bothered to read the WIKI about how to use the FREE firmware. So you call us pricks because we won't write a novel with step by step instructions tailored especially for you when it is all in the WIKI.  Real nice!!

BTW, the page I linked has all of the information you need in excruciating detail.

chamnit, 2018-12-19:
@109JB : To be fair, no one answered why there was change or what to do. Only just a link. I think the missing piece here is that his EEPROM was reset when he upgraded and didn't realize it would do that. (There are a few warning about this in various places.)