gnea/grbl #582: Feature request - Fault input to Grbl

109JB, 2018-12-21:
I have a suggestion for a feature that could be implemented in the future. For now I have a work-around that I will also describe. 

So, here is the issue. I am currently converting a new milling machine to CNC and I am using closed-loop stepper drivers. The loop is closed between the drive and the motor, so to Grbl or any other controller, all that is needed is step and direction like any other stepper.  However, the drivers also have a pin that can notify the controller of a following error fault.  If a drive faults it stops motion on that motor, but the other axes can still go on merrily, so the fault has to be communicated back to the controller so that the controller can stop the other axes.

Right now am going to implement this using the Safety Door feature that Grbl already has and I think that it should work OK, because it will pause motion on all axes, but it could be better which I will discuss below. The only thing I plan to change eventually is the messages reported back to the computer (ie: "Door:" to "Fault:", "Check Door" to "Drive Fault", etc.)

These closed loop drives are becoming much more affordable and are going to become more commonplace.  I think for this reason it would be prudent to have a dedicated pin/pins for this fault detection.

As for how to implement this in a better way than the safety door pin, there are a few things I think are lacking in the above implementation.

1.  Tying all of the drive fault pins into one input to Grbl works but means that the operator only knows a drive faulted but doesn't know which one. The drives itself will display a fault condition, but once the machine is done the drives will be in a cabinet and would require opening the cabinet to see which drive faulted. Knowing which drive faulted is important because if a drive faults there is a problem and knowing which axis to look at will help with solving the issue that caused the fault in the first place.

2. The Safety Door feature is really a feed-hold type of deal. When there is a drive fault what really should happen is an Alarm should be issued and if homing is enabled, then homing should be required to resume.  However, using the safety door does work for the immediate need to halt motion on all axes.

3. The messages sent from Grbl back to the user interface should reflect what is actually happening. For example, if there are individual fault pins for each axis then an alarm message like "ALARM: Drive Fault Y" with the status report showing machine state as "Fault: Y" might be possibilities. Other messages could be implemented accordingly.

4. Using the safety door input for this means it isn't available for the door anymore. For me that isn't a problem because I don't plan to have that anyway, but I know other do need the safety door input.

Thanks.

John Brannen

MeJasonT, 2019-02-17:
Hey buddy
If for example you had built a pendant, the Emergancy Stop would be wired to the reset pin.  This immediately and completely disables any actions and locks GRBL.  As with most professional machines there will be a number of switches which indicate an alarm condition (often controlled by safety relays but more often than not breaking a continues circuit to the power contractors).  For GRBL the pendant ES (and a proper door switch any i have seen would kill the machine if an idiot opened it while running) be a perfect example of the safety circuit, I have been looking at installing a door switch to my machine and have commented on a couple of posts regarding its configuration.  It came as a surprise to me that by default it is coded to be a closed contact when the door opens - completely reverse of an industrial electrical safety circuit where the opening of a circuit takes the control voltage away from any switches or contactors thus rendering the machine disabled.
But actually thinking abut GRBL it makes perfect sense as hitting the reset button on the pendant or opening the door connects that switch to ground.  so what we have is a parallel circuit instead or a series one , where in an industrial situation each switch is ANDed together to keep the circuit live with GRBL we have the advantage of using the switches in an OR state.  As GRBL is using a micro controller and not relay logic, wiring in a circuit which cuts the power to isolate the machine would also kill the board, loose the settings etc.  Its a case of understanding what we have to work with and the reasoning behind what we want to achieve.  If a drive fails whilst running a job then i would argue that was critical enough to hit a stop button, So with that in mind the most logical thing would be to take the outputs from all the motor alarms and wire them through relays or other circuitry so that they trigger the reset pin to ground.  I believe the End stops operate in a similar way all be it they have there own pins.

I keep forgetting you guys are on bigger and better processors now - i'm still there hanging on to my UNO.  On the mega any amount of input pins can be assigned and there conditions accumulated into on alarm message, s rater than send ALARM:mot1,mot2 etc we simply opted for ALARM (a Binary coded decimal number which is bit weighted for which pin is triggered) Eg, ALARM:4 which is actually 00100000 therefore device 4 has triggered.  it will become obvious which motor has died when you restart and send HOME command, it will be dead.  On the other hand if the motor overheats and alarms you will end up chasing your tail all day, sods law says when you restart he machine the alarm will clear only to overheat again.

109JB, 2019-02-17:
I do not like the reset pin for a drive fault for several reasons.

A drive fault is something that needs to be conveyed back to the control software and hence the user, but it also has to be handled in a particular manner. A drive fault means that an axis drive has exceeded the position error following limit.  That is all it means.  It does not mean that a critical emergency condition exists. Handling it as an emergency stop, which a reset would do, could cause more problems.  What is needed for a drive fault is a controlled stop that leaves the spindle running so as not to break the tool or stall the tool during cutting. 

Ideally, and the reason I started this thread, we would have a fault input that could be customized in its response. For example, I think the ideal thing would be to stop the axes, and then once the axes stop, delay for a few seconds and then shut off the spindle and coolant. Perhaps the user could have the ability to customize the delay time between axis stop and spindle stop, and whether to stop spindle and/or coolant.

I don't feel that a reset can't be used for this because as soon as a reset is invoked the spindle stops, and the drive motors stop in an uncontrolled manner. Also, the controller software needs to know the difference between a drive fault and a reset. On this machine I plan to use the reset pin wired to an external button, so there would be no way for the control software to know the difference between me pushing the button and a drive fault. Since I am not using the safety door feature on this machine I can use this pin for the drive fault input, and it helps that the safety door function closely matches what I feel should be done in a drive fault scenario, but it could be better if it were customizable.

While having multiple pins for each axis so you know which axis has the problem could be helpful for other situations, I don't need this feature because the drives I have display the fault, so I can know which axis is faulting by looking at the drives. The fault on the drive cannot be cleared until the drive is power cycled, so would not be a phantom fault.

MeJasonT, 2019-02-17:
I understand your point regarding regarding the spindle stop and saving the tool, I think in reality the motor will have lost steps and grbl will have sent commands which the motor may have the tendency of trying to carry out.  I say this from the experience i have just had where i stalled the whole machine my Z motor does not have enough holding torque to hold the tool at height, this allowed the tool to dig in - luckily my spindle is belt driven so it slipped when the tool snarled up.  All the motors ended up stalling from being jammed but also overheat protection.  The tool is in one piece all be it a lot blunter.  I think if you miss any steps not considering it as an emergency is not a good move (no pun intended).  However the safe parking makes more sense - an alarm would need to be triggered as the coordinate system will now be out of kilter (we missed steps).  In most cases unless jammed the spindle would go into free running (stop condition) and would be ramping down to a stop, depending on DC brake injection if used.
With my VFD I can either have a rapid stop or a free running ramp down, not both available at the same time, programmed either or. Ok its cheap and old, must be at least 12 years by now. 
 An additional pin would make interfacing with the controller a lot easier
The next question would be should the door stop rapidly shut off the spindle instead of ramping down.  I guess in the real world a machine is idiot proof and hits the brakes on the spindle if an attempt is made to open the door.

109JB, 2019-02-20:
>I think in reality the motor will have lost steps and grbl will have sent commands which the motor may have the tendency of trying to carry out.

The motor in fault mode will not be trying anything as it will be disabled and the fault detection portion of the GUI program is there to shut everything else down

>I think if you miss any steps not considering it as an emergency is not a good move (no pun intended).

We can agree to disagree.  My drives will fault with a following error of 0.05".  One axis getting 0.05" out of position does not by itself mean an emergency and my whole purpose in implementing fault detection is to stop things before it rises to the level of an emergency. 

Imagine drilling a hole and having a bit too high a feed rate. Allowing the spindle to run after the fault would allow the drill bit to clean the hole with no damage to the part, tool, or machine.  An immediate stop would leave the tool buried in the part with the cutting edges of the drill still embedded into the metal. Depending on the conditions, this could result in a broken tool, or a ruined part.  This is only one scenario, but I could come up with many situation in which allowing a controlled stop would be better.

MeJasonT, 2019-02-20:
Its a good point you made and certainly has merit in the debate.  Its clear that a user of grbl would benefit from such a safety feature and we have certainly made good head way into the pro's and cons regarding both of our stances on what is intended by such a feature.  I'm sure our joint effort will be well received when it comes to modifying the code and adding additional features.
I am in full support of your proposal for this feature to be added.