gnea/grbl #584: Acceleration error frizzed arduino

Leoavilacr, 2018-12-23:
I am using [VER:1.1f.20170801:] on an Arduino NANO clone.

While I was doing some test and by mistake I change some parameters to 

$100 = 100 'steps/mm'
$110 = 50000 'mm/min'
$120=1000

I ran a G0X300 command and the stepper stalled. I know I'm exceeding 30kHz step rate so I try to change them back but I can't access the data in the EEPROM, I used several GUI in both mac and pc to send a reset command or to flash fresh grbl into it but it doesn't do anything.

When try to connect to grbl with laserweb it says firmware does not supported, with UGS it says it's grbl is still loading and LightBurn says it's busy and won't get any command. Xloader and HexLoader did not flash the arduino

What could i do to fix this?

109JB, 2018-12-24:
Use a serial monitor program and then send "$RST=*" to Grbl to restore all EEPROM data. Or, upload Grbl again using the Arduino IDE.

chamnit, 2018-12-24:
You can try wiping your EEPROM. I think there is an example sketch that will do that. You just have to alter the byte range to 2 KB.

chamnit, 2018-12-25:
The Arduino Nano sometimes doesn't flash correctly in my experience. If you reset the Arduino within a second of the Arduino IDE flashing the Nano, it should connect and complete the flash.

closed 2019-02-19 by chamnit