gnea/grbl #585: After homing machine, G28 triggers limit switch

David-Hari, 2018-12-24:
I have an [XYZ-Carve](http://xyzcarve.3dtek.xyz/), which is similar to the XCarve. I use the [UGS Platform](https://winder.github.io/ugs_website/) software to send G-code to the machine.
When using the G28 command to go back to zero, the tool travels all the way along the Z axis and hits the limit switch.

I have set the firmware settings to the following:

```
Max travel (mm):
  X: 1000
  Y: 1000
  Z: 200
Homing pull-off: 1 mm
```

The X & Y limit switches are at the bottom left of the machine, and Z is at the top.

After homing, UGS displays the following position (mm, first number is work pos, second number is absolute machine pos):
```
  X:  0.00      -999.00
  Y:  0.00      -999.00
  Z:  199.00    -1.00
```
I read somewhere that the negative co-ordinates are normal for CNCs, so although it seems odd to me, I am not concerned about that.

Issuing the following command,
```
  G90 G0 X0 Y0
  G90 G0 Z0
```
takes the machine to
```
  X:  0.00     -999.00
  Y:  0.00     -999.00
  Z:  0.00     -200.00
```
which is bottom left of X/Y axis and Z all the way down.

And the following,
```
G91 G28 X0 Y0
```

takes it out to
```
  X:  999.00     0.00
  Y:  999.00     0.00
```

However, when moving the Z axis to zero,
```
  G91 G28 Z0
```

it hits the limit switch and triggers an alarm and immediately stops.
This is because it thinks that Z 0 is right on the limit, instead of being just shy of hitting it.

I imagine the correct thing for the machine to do after homing is to set machine Z to 0 instead of -1. Hence, the zero position is where the tool stops after homing (which is 1 mm away from the limit switch).

Please tell me if this is a bug in the grbl code, a problem with the XYZ-Carve, or have I just done something wrong?

chamnit, 2018-12-24:
Read [this FAQ](
https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#why-is-grbl-in-all-negative-coordinates-after-homing-or-it-so-annoying-and-not-what-im-used-to). Also please review how work coordinate systems work. LinuxCNC has excellent help pages.

closed 2018-12-24 by chamnit

David-Hari, 2018-12-25:
Thanks, that was useful information, but it doesn't really answer my question.

I do have my limit switches wires normally-open. The homing works just fine. It's just where it defined Z 0 to be that is the problem.

neilferreri, 2018-12-25:
@david-hari What are you expecting that command to do? Do you know your G28 offsets? My guess is you have not set any, so they default to 0,0,0 (away from your homing switches). Without more details, I'm not sure why x and y aren't on their switches as well.
If you enable HOMING_FORCE_SET_ORIGIN, you can set your machine zero at a pull off distance from the switches. 
Again, though, what are you trying to accomplish with the G91 G28 Z0 (move Z to its G28 offset)?

David-Hari, 2018-12-25:
> Do you know your G28 offsets?

What do you mean by that? I thought G28 was supposed to go the machine’s reference (or zero) position. That's what I read [here](https://www.cnccookbook.com/g28-g-code-cnc-return-reference-position/) and [here](https://www.cncci.com/resources/tips/how%20g28%20works.htm).

My problem is essentially that Z 0 triggers the limit switch, which I don't believe it should do. Moving to 0,0,0 should be a safe operation. The limit should only be triggered if it moves slightly past 0. Does that make sense?

neilferreri, 2018-12-25:
Through the console in UGS, send 
$#
That will show your offsets.
You can define your G28 position by sending a G28.1 command when the machine is in the location you like.
Grbl uses G28 as a user defined position. I use it for a tool change/probe location and I use G30 (works the same) for a get out of the way position.
As Sonny mentioned, linuxCnc is the place to get your info.
Here's a pertinent link: http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g28-g28.1

David-Hari, 2018-12-25:
Thanks, that makes a bit more sense now.

I'm not sure the XYZ-Carve's firmware supports decimal g-codes (I think it complained once when I tried it), but I'll give it a go.

Do you think UGS should be using G28.1 immediately after homing, so that G28 will return to that point? I can always make a macro, but I would have thought it would be part of the homing process.
Perhaps I should report it on their Github page.

neilferreri, 2018-12-25:
No. You don't want your UI changing those offsets. Everything is working. You set your G28 where you want it. I'm assuming since you posted here that the machine is using GRBL. It should accept G28.1
If you want to go back to machine zero, just home the machine.

David-Hari, 2018-12-25:
> If you want to go back to machine zero, just home the machine.

That's the problem - machine zero is beyond the limit. Well, _at_ the limit, which causes it to trip the switch. Are you saying this is ok? That it is normal for it to do that, and that I need to manually change it using G28.1? I don't understand.

neilferreri, 2018-12-25:
Machine zero is the limit switch by default. You won't be using machine coordinates for your work.
Were you able to see your current offsets? $#

David-Hari, 2018-12-25:
No, I don't use machine coordinates for my work. But the G28 command does. Since you say I can change where G28 goes to then I guess my problem is solved.

David-Hari, 2018-12-26:
I should have mentioned that the g-code is generated by Fusion 360, which by default does a G28 at the beginning and end of a job. I can disable that functionality, but the purpose of asking here was to determine how I can use it successfully. As neilferreri said I can set the position that G28 goes to, then I will try that.

neilferreri, 2018-12-26:
@David-Hari I use Fusion 360 as well, but I use a custom post. Like I said, I like using G28 for a tool change location. That command (G28 G91 Z0) in the default Grbl post is the reason for several threads on the Inventables forums. I'd recommend joining there.

ramimehyar, 2022-10-23:
Hello David, 

Did you find a solution to your issue, ever since I updated my FW, the Grbl homes to the limits, pulls of 1mm and then sets all home coordinates at -1.000, -1.000, -1.000

langwadt, 2022-10-23:
rebuild with HOMING_FORCE_SET_ORIGIN defined, that will set the machine coordinates to 0,0,0 after pull off