gnea/grbl #588: Limit switches trigger on RPi and not on laptop

aleslukek, 2019-01-03:
I have a very strange problem. I have Arduino Uno (original) connected to the Raspberry pi and running UGS (universal gcode sender) on it.
Hardware wise I have limit switches (normally closed) connected to Arduino. I have low pass filters with operation amplifier on it. It is set so, that when limit is triggered the op amp sends 5V to Arduino, if limits are not triggered, it sends 0V. Before the signal goes to op amp it is filtered with low pass filter, as described in grbl wiki.
Recently I also changed all cables for limit switches. I used UTP cat 6 (twisted pairs with shielding connected to earth) to further reduce interference. Shielded cable is from switches to filter and then again from filter to Arduino. So the signal should not get interfered in the last few centimetres.

However I can't seem to get rid of random limit switch triggering when running from RPi. But if I connect Arduino to my laptop (MacBook) the limits don't randomly trigger when jogging - when motors move - anymore. I tried to reduce motors acceleration but it helped so little it's not worth bothering.
Arduino is powered from USB, RPi is powered from 50W 5V industrial power supply. Op amp is powered directly from the same power supply.

tldr: I don't know why limits trigger only when running from Raspberry pi and work ok when running from laptop.

I would be grateful for your ideas.
Aleš

bobbynohairs, 2019-01-03:
Is the power supply a cheap Chinese? Do you have access to a oscilloscope? I would check 5V rail for interference during fault conditions. Try to keep supply cables between power supply and raspberry pi, as short as possible. My first step would be use a quality 3A supply, that is recommended by the pi foundation, to power the pi on it's own, to see if that helps. I too use shielded cat cable for limit switches/sensors. I have found that good practice is to earth shielding, radially. That is earth only one end of cables and terminate to a common earth point on machine chassis. Earth loops are a known cause of interference on low level signal cables. Earthing of shielding would seem to be not your problem, as you say everything's ok when using the mac. Another golden rule is, what goes on in USB, 
stays in USB. Try to not share common power supplies with USB, unless you can really help it, or you will need additional filtering of supply rails. I have long since ditched direct dc switching for limit switches. I prefer to connect the switches via opto-couplers. Hope this is of some help.
Bobbynohairs

aleslukek, 2019-01-03:
Power supply was bought in Europe, but chances are that was made in China. Eventhough it was supposed to be a better quality. 

Cable shielding is connected to earth only at one side. Obviously.

You have given me an idea, to try to power the Arduino separately. Directly to power supply and will try powered usb hub just to see if it makes a difference.
I will try it this weekend and report back. Will also probe with oscilloscope.

I wasn't thinking into changing to opto-coupler, since I would assume the issue would persist on laptop as well not just RPi - if it was a filtering issue. To be truthful my filter (lpf with op amp) worked great even before changing into shielded cables. Without filters limit switches did not work. 
I changed to shielded cables thinking it would remove thud issue with RPi, but it didn't. So I highly doubt opto-couplers would make this go away. I have been mistaken in the past, so if nothing helps I will try them, as a last resort.

Problem is that I don't want to spend an extra couple of hundred euros just to buy new a computer once whole set up just to deal with this stupid random triggering. I used RPi with touchscreen just because it is cheap in dusty workshop and works great if everything goes as planned.

bobbynohairs, 2019-01-03:
I'd do one thing at a time. If your Arduino and machine set up is ok with the mac, I would still trouble shoot the Pi side of things. Another thought, are you using the same usb cable to connect the mac as the pi? Honestly I'd get a recommended 3A supply, for fault finding. They are not expensive. I always keep one for fault finding. Cheap noisy switch mode supplies have caused me extreme grief over the years.

carneeki, 2019-01-03:
Double check the grounding too.

If the ground is not common to all devices, and something mysteriously
floats, that could cause false triggers.

I'd also consider a second LPF very close to the Arduino inputs, but if it
was fine with a PC, perhaps some better power filtering and the problem
will go away? RF suppression beads close to the pi maybe? (Or wrap the
cable around a toroid a few times).

aleslukek, 2019-01-03:
I will check the grounding, thank you for a suggestion.
Router wires are routed elsewhere. So I can take it off easily if needed elsewhere. Together there are limit switches and servo wires. I also routed wires for TTL laser if I decide to purchase it in the future (so no, laser is not yet installed). And yes, those are shielded from others as well. Once I dug all wires out I may put some extra in case I decide to use them some day.

bobbynohairs, 2019-01-03:
I run my spindle wires inside earthed stainless steel "Goose-neck " flexible conduit. I also removed VFD internals from plastic box and installed in a earthed die-cast aluminium box. I have an inductive pick-up hitched to my oscilloscope. Only about 10% of original radiation before modification.

aleslukek, 2019-01-06:
Just to report back.

In short it was a ground issue. I rewired the ground for an op amp and Arduino again and it sorted the problem out. Arduino is now powered directly from power supply as well as an usb.

I probed the limits and 5V power supply.
Signal coming from limit switches is flat as a pancake. Now even before coming to a filter, it doesn't move a bit when motors start, stop, move or idle. After filtering the signal is just a bit stronger since it is amplified. So all in all, changing to a shielded wires helped a bit, but did not fix a problem I was having.

Power supply voltage is also flat on motor movement.

So it wasn't a limit switch wiring issue, but an issue with a common ground for some reason in a combination with RPi and Arduino. Now limit switches don't trigger automatically. 

Now all that is left is to figure why there is "Error while processing a response", other similar errors and stoppages without any error going on on RPi and not on Laptop in Universal Gcode Sender. Well, at least alarms don't trigger anymore. Thank you all for your ideas, they helped :)

closed 2019-01-06 by aleslukek