gnea/grbl #590: Please add package_index.json for Arduino IDE

Harvie, 2019-01-09:
Hello,
can you please publish `package_index.json` file somewhere on github, so i can easily add GRBL to my Arduino IDE and have it automaticaly updated:

Eg.: Adafruit has this URL:

`https://adafruit.github.io/arduino-board-index/package_adafruit_index.json`

You add it:

[image]

And then you can easily download and install libraries and sketches from their repository:

[image]

I think this would be nice way to distribute and udate GRBL, since it already can be installed through Arduino IDE... But now we have to manualy download it.

per1234, 2019-01-10:
You're confusing two different things.

[package_index.json](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x-package_index.json-format-specification) is used to allow hardware packages to be installed via Boards Manager (Tools > Board > Boards Manager). This is the feature related to **File > Preferences > Additional Boards Manager URLs**. grbl is a library, not a hardware package so this is not relevant.

To make it easy to install and update libraries, the Arduino IDE has a somewhat similar feature: Library Manager (your second screenshot). A really nice feature of Library Manager is that it notifies the user when new versions are available of any library they have installed. The process of getting a library into Library Manager is different and much easier than adding Boards Manager installation support to a hardware package.

Here is what would need to be done in order for grbl to be [installable](https://www.arduino.cc/en/guide/libraries#toc3)/updatable via the Arduino IDE's Library Manager:

1. Move the library to the root of the repository. This will also permit easy installation of the GitHub .zip files via the Arduino IDE's **Sketch > Include Library > Add .ZIP Library...**
2. Add a library.properties metadata file, compliant with the [specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format). (https://github.com/gnea/grbl/issues/350)
3. Create a tag or release that matches the `version` value in library.properties.
4. Open an issue report in the arduino/Arduino repository requesting addition in the Arduino Library Manager index.

Once the grbl repository is added to the Library Manager index, all new tags will automatically be picked up within an hour so there will be no further work required by this process.

If this is something the admins are receptive to, I'm happy to submit pull requests for (1), (2), and do (4). I am not able to do (3) due to not having those privileges.

Reference:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Harvie, 2019-01-10:
> You're confusing two different things.

I guess you are right. But note that board manager can also install libraries (as you can bundle them to board definition) and can do so even without being acked by arduino guys (unlike library manager).

But i guess if it's possible to get ACKed by arduino, it would make more sense to use library manager. (Even when GRBL is not really a library).

per1234, 2019-01-10:
>But note that board manager can also install libraries (as you can bundle them to board definition)

It's true that you could turn grbl into a hardware package with grbl as a bundled library, but libraries bundled with hardware packages are only accessible when a board of that package is selected from the **Tools > Board** so this would limit the number of boards grbl could be use with to only the ones that have definitions in the grbl package. This would be a very hacky workaround that would require a lot of work, make installation more difficult (due to the extra step of adding the Boards Manager URL), and provide no benefit.

>if it's possible to get ACKed by arduino

After submitting the issue requesting inclusion in the Library Manager index, it should take less than 10 days for Arduino to do it. I will personally guarantee this. This is a routine process that has been done for close to 2000 libraries already.

>Even when GRBL is not really a library

GRBL is in the form of an Arduino library.

>In that case there will be no need to add new index URL, so it will be more user friendly. Just open stock arduino, search for grbl and you are good to go.

I agree.

Harvie, 2019-01-10:
> GRBL is in the form of an Arduino library.

I know. It's in form of library, but it is not a library. So lets see how they will aproach such case. Personaly i hope they will accept it, since GRBL is widely used. Maybe this will even make them consider add "firmware/sketch" categhory to their library. To indicate, it's ready made firmware rather than library used to build custom sketches.

I can imagine that they can add such option for projects like grbl right in this menu:

[image]

Harvie, 2019-01-10:
> version - version of the library. Version should be semver compliant. 1.2.0 is correct; 1.2 is accepted; r5, 003, 1.1c are invalid

This is not very good for current grbl versioning scheme, however according to https://semver.org/ it should be possible to use following version format in arduino:

1.1-f, 1.1-g, etc...

Harvie, 2019-05-06:
Any news on this one? This should be VERY easy to implement...

Harvie, 2020-02-23:
Can i make a pull request? It will need following renaming:

grbl/examples --> examples
grbl --> src