gnea/grbl #594: PNP NC Inductive Endstop

raenji-sk, 2019-01-29:
Hi, is it possible to use PNP NC Inductive endstops with GRBL?
I have some laying around, but cant seem to get them to work.
I am using an UNO with CNCDesign breakout board - 
[https://ukcnc.net/index.php?route=product/product&path=69&product_id=139](url)
and bCNC for control.
The sensors are [https://www.tme.eu/sk/details/s18-05p-2/indukcne-senzory-obdlznikove/highly/](url)
The sensor LED is showing that the metal is detected but grbl does not respond. 
I tried inverting the pins $5 - parameter with no effect.
Is it possible to make it work or does it need NPN sensors?

HuubBuis, 2019-01-29:
The problem using these switches is the voltage at the output. I will reach the level of the supply voltage and destroy the arduino. You can solve this by adding an optocoupler or a level converter [https://www.digikey.nl/product-detail/en/sparkfun-electronics/BOB-12009/1568-1209-ND/5673795?utm_adgroup=Evaluation+and+Demonstration+Boards+and+Kits&mkwid=s&pcrid=285092047603&pkw=&pmt=&pdv=c&gclid=EAIaIQobChMI2qKEnKuT4AIVh-J3Ch0e3wY-EAQYASABEgLTVPD_BwE](url) . Your shield has optocouplers on the limit switches so you can connect the Inductive sensor to thes limit switches. For this to work, the Inductive sensor GND must also be connected to one of the optocoupler pins. The Inductive sensor uses a high supply voltage. To limit the current through the optocoupler, you have to add a series resistor. Contact the manufacturer of the shield for the proper resistor value and the proper way of connecting your inductive sensor.

raenji-sk, 2019-01-29:
Yes, the optocouplers were one of the reasons I bought it. I am using a 12V PSU for the sensors. I will try contacting them about this issue. The schematic shows direct connection to the shield and a 12-24V power for the sensors, but they do not mention whether it is for PNP or NPN.

raenji-sk, 2019-01-29:
So they have to be NPN, PNP do not work with this shield.

closed 2019-01-29 by raenji-sk

HuubBuis, 2019-01-29:
Can you post the schematic, NPN and PNP can work if the input is an optocoupler!

MeJasonT, 2019-01-29:
Try using the proximity sensor at 5V you may be presently surprised.  Some transducers will accept and operate at voltages much lower than that rated.  May not work with all but certainly did with the cheap china imports I purchased.  Its also possible to use old logic ICs to such as 4500 series or 7400 series ICs to convert your transducers 12v signal to 5v logic output - if the IC is supplied with 5v on its power rails.  most Logic ICs in these series are 18v input tolerant.  You also have the beauty of changing logic states, for example you need to invert the signal then select a logic device which has a NOT (inverted) output.

[https://en.wikipedia.org/wiki/List_of_7400-series_integrated_circuits](url)

Eg a 7407 buffer or a 7404 inverter IC

langwadt, 2019-01-30:
I'd like to see a datasheet for 4500 series or 7400 series ICs that allow 18V input with 5V supply.

for 10-30V PNP sensor to logic input just use two resistors as voltage divider, one of the resistors with be needed anyway as a pull down

raenji-sk, 2019-01-30:
HubBuis, I do not have the schematic of the shield. I got this reply from the manufacturer: "The opto isolator has 12V connected to it via a resistor all the time and the input to turn that on has to be a GND(0v) so NPN will switch 0V for you when activated." So I guess it is the exact opposite of what you wrote.
I could maybe connect it via relays, but I do not think it is the cleanest solution. I will stick with classic mechanical endstops for now, maybe order some cheap NPNs from China in the future.

HuubBuis, 2019-01-30:
@raenji-sk 
Your sensor has a 12 Volt output when not activated and 0 Volt when activated. That is typical for a PNP NC sensor and that is the right sensor to use. When the sensor loses power or the cable breaks it will be the same as an activation and GRBL will stop! This is the proper way to use safety switches!

You can connect the 12 volt sensor output using a resistor of 1500..2700 ohm (1k5 .. 2k7 Ω). This will limit the current fed into the optocoupler input of the shield.  

I am not sure how to connect the sensor to the shield. There are 3 pins on the shield ZL-+, ZL-I, ZL--. **I guess** **ZL-+** sensor output, **ZL--** sensor gnd and **ZL-I** the input on the arduino (output of the optocoupler).
Try to find out the meaning of these labels and you can use this sensor.

MeJasonT, 2019-01-30:
langwadt
Sorry I meant to put 4000 series ICs it was late and I was half asleep
[https://assets.nexperia.com/documents/data-sheet/HEF4050B.pdf](url)

> The HEF4050B provides six non-inverting buffers with high current output capability
suitable for driving TTL or high capacitive loads. Since input voltages in excess of the
buffers’ supply voltage are permitted, the buffers may also be used to convert logic levels
of up to 15 V to standard TTL levels. >

I have corrected post above

langwadt, 2019-01-30:
@MeJasonT as long as the series resistor limits the input current through to ~1mA is is still safe

MeJasonT, 2019-01-30:
I've just had an Epiphany not as wet as a Eureka moment.  If you have electronics skills, it may be possible to change a resistor on the shield which is the input to the opto isolator/optocoupler.  It is basically an led transmitting the signal to a photo diode.
or as @HuubBuis mentioned earlier connect the input in series with a resistor - 1.5k should do it.
you can calculate the current limiting resistor using the calculator here
[http://ledcalc.com/](url)
I used 12v supply
2v across the LED (typical value)
10mA which is very conservative for an LED
and obviously 1 led
which comes out to 1.2k ohms ( this is a nominal value, ie you can buy one) going slightly higher is just belt and braces (a safety margin).

raenji-sk, 2019-01-30:
@HuubBuis Thanks for recommending the resistor solution. After some measuring I found out that the load is small and can't get the opto to activate. Adding a 1k8 resistor between the signal and gnd wires on the sensor fixed it. Now the signal gets to complete 0V and "triggers" the opto and Grbl receives the change and all is working as desired.