gnea/grbl #595: Stall on milling

lolowl-coder, 2019-01-30:
Hi. I've encountered **communication issues** with Arduino Uno r3 clone board, equipped with 16u2 chip.
Suddenly, my Arduino and GCode streaming software freeze when milling.
What I tried:
- GRBL v1.1 (grbl-1.1f.20170801), v0.9j
- Firmware update for 16u2 as recommended [here](https://github.com/gnea/grbl/wiki/Known-Issues#usb-to-serial-transmission-errors)
- Baud rates (9600, 115200, 230400)
- UniversalGCodeSender, bCNC
- Different usb cables:
   - 0.3 m, shielded supplied with arduino uno
   - 1.35 m, shielded
   - 1 m, diy cable from mobile phone
- GCode-file ~10 Mb [1001.nc](https://drive.google.com/open?id=1c9gv71UXO6z14D3SdkL-fkMe9hZ5x2ke)

After these steps problem still remains.
How to fight this? Any help will be appreciated.

HuubBuis, 2019-01-30:
Consider using Bluetooth or WiFi in stead of USB for elimination ground loops and interference from the high current stepper controllers. I use WiFi for about 2 years and it has never let me down!

lolowl-coder, 2019-01-30:
@HuubBuis, thank you for fast response.
Is this [board](https://store.arduino.cc/arduino-uno-wifi-rev2) suitable for your idea?
Should I change something in firmware?

viewsat, 2019-01-31:
@lolowl-coder 
You can use Grbl-Panel to send g-code.
Line 11 of 1001.nc is an unsupported command
[image]

swarfer, 2019-01-31:
I have just run your job through my clone Uno with CH340 chip in check mode with success.
all 490000 lines of it, but I did edit out the toolchange line (maybe you should use a different Fusion post, one really made for GRBL :-)  https://github.com/swarfer/GRBL-Post-Processor

What does this mean?  
you say 'suddenly it stops' which I take to mean it used to work perfectly but now it does not.  This indicates something has changed, maybe the USB driver got updated?  or the Uno got hot once too often and there is now a dry joint.   MAybe point a fan at it while it runs and see if that helps.  If so then the problem is heat related.
is the PC staying on during the job?  maybe it is trying to sleep (yes the screensaver/power setting can change during a Windows update)

MeJasonT, 2019-01-31:
Does reducing the status query period on your GUI help, you can have too much data.  200ms is certainly fast enough for human monitoring.  I found the the GUI Candle [https://github.com/Denvi/Candle](url) was set by default to 40ms, so that's 25 responses per second with a string 70 characters long being reported back from GRBL.  It cause a few data dropouts for me until i reduced it.  GRBL keeps track of its location the response messages are just for the users reference/display purposes so you can see what the machine thinks it is doing.

lolowl-coder, 2019-01-31:
@swarfer,
> What does this mean?
It means my machine starts normally and after some (random) time movement slows down (may be without slowdown) and steppers stops to move.
> MAybe point a fan at it while it runs and see if that helps.
It's not overheating. Tried with cooler 80x80 mm.
> is the PC staying on during the job? maybe it is trying to sleep (yes the screensaver/power setting can change during a Windows update)
Yes, PC is on and normally operated (not trying to sleep, etc.)

Next thing i tried - process same file (in first post) with spindle disabled.
File was successfully processed.
I think the cause of problem was interferences from spindle (500w, brushed, aliexpress).
Does anyone know how to ensure in this?

I like wifi-approach, as @HuubBuis recommends. I'll try it.

I'm also interesting in how grbl behaves if it can't confirm, that gcode line is successfully processed?
Is such case a deadlock (no confirmation, no next GCode line, freeze)?

HuubBuis, 2019-01-31:
> I think the cause of problem was interferences from spindle (500w, brushed, aliexpress).
Does anyone know how to ensure in this?>

- If the brushes wear out, they will make less contact and  generate more interference
- If the grounding of the spindle changes (more, less, none) interference also changes. Check for lose bolts on wiring and mounting or oxidation
- Did you change the wall outlet
- Did you change, add or remove other equipment (Zero probe)

nsiatras, 2019-02-14:
I experience the same problem. Do you connect the Arduino GND pin to A/C Earth ?

HuubBuis, 2019-02-14:
The Arduino GND pin should not be connected to AC/Earth. This is basically the same for every electronic device. The problem is that if you connect multiple electronic devices, earth current/spike go's the easiest way, which could be the electronics GND connected to ground. It could not only lead to erratic behavior but also destroy the electronics.

nsiatras, 2019-02-14:
I noticed that by connecting the USB to the arduino the GND pin is
connected to AC/Earth.

HuubBuis, 2019-02-14:
I checked the Arduino Uno R3 schematic and yes, the USB shielding is connected to the Arduino GND by a small coil. This is probably done by a lack of a shielding Pin. This could results in ground loops and is in my opinion a design failure. I also have checked my own processor boards and I have made the same mistake (oeps)!
Shielding can be very useful. The basic rule is to connect shielding only at one side, the most sensitive side and connect the shielding to earth.
The easiest way to do this, is to cut the shielding at the Arduino end of the cable. This way you have a proper shielding and if your computer has a earth outlet,  the shielding works even better.

nsiatras, 2019-02-14:
I don't believe is a design failure. I just checked an STM32 Nucleo board.
Same thing here. The GND pin is connected to A/C Earth as you connect the
board to the computer

109JB, 2019-02-14:
According to this document http://www.ti.com/sc/docs/apps/msp/intrface/usb/emitest.pdf the USB shield should be tied to the ground plane. See section 5.4 where it states:

>Full speed devices use a shielded cable which requires that the connector shell be tied to the ground plane.
It is important to note that a ground plane does not behave like an equipotential surface at high
frequencies.

HuubBuis, 2019-02-14:
@109JB 
According to this documentation you are right and this document wasn't written by dummy's. 

The fact remains that connecting cable shields on both sides results in ground loops, ground loops results in troubles and for proper shielding only one side of the shield needs to be connected.  I don't know what is wise to do, but without changes there is not improvement.  

After reading this document I still think it is worth a try to use a USB cable having the shielding cut at one side.

langwadt, 2019-02-14:
cutting the shield won't do much since there is a ground wire in parallel with it, USB needs ground

HuubBuis, 2019-02-14:
The problem are not the 2 USB ground connections, but the earth current that could and will flow when earth grounds of different electronics are connected and this connection isn't a star topologie (single point central earth)

langwadt, 2019-02-14:
but if the PC grounds the shield cutting the shield will not break the loop since there's still the ground wire to break the possible loop you need an USB isolator or use a laptop running on battery

n0th1n6, 2019-02-15:
I've been scratching my head about this for a week, almost gave up pcb
milling. When running the gcode with spindle also running but not cutting
anything, the job compleles without a stall. As soon as I let the tool
touch the copper of the pcb, it will randomly stall.

So i did experiment

1. Use a seperate line. Different breaker so the path from arduino socket
to the spindle socket is longer. It did improve the situation but will
still stall. Quite frustrating coz it sometimes happen at around 90%.

2. Added choke to the USB cable. Did not solve the issue.

3. Added choke on the spindle wire. It significantly improved the
situation. I can even plug the controller board and spindle in same line.

4. I picked up a good quality but short (400mm) usb cable. Now I have
almost forgotten that this ever happened if not for this thread.

In summary, my guess is that the stall is caused by a high freq noise
produced by the motor that travel to the controller board and messes the
usb timing. The choke on the spindle prevents the noise from reaching the
controller board.

Also, just a precaution, avoid plugging a usb device that draw high current
(phone, mass storage device, etc) during milling. A surge in current drawn
from the usb can shut down the usb controller.

HuubBuis, 2019-02-15:
The Arduino has no connection to earth ground so ground currents can't travel trough the arduino to reach earth ground.

If at the PC side, the GND is connected to the Earth ground (probably yes), then the shortest route for the earth currents to travel will not be via the Arduino USB back and forth but via the PSU (power outlet) connection.

n0th1n6, 2019-02-15:
That might be the case. Still my #3 and #4 applies as the choke is on the
spindle.

dcavataio, 2019-04-01:
Same issue, the choke on the spindle took care of the stall.

matrix-wrangler, 2019-09-16:
@n0th1n6 @dcavataio facing same issue here, please let us know the parameters of choke that fixed the problem and specs of your spindles.

n0th1n6, 2019-09-17:
@matrix-wrangler, choke did not solve the issue for me. Following solved this issue 100%

1. Use high quality USB cable. The shorter the cable the better.
2. Using dremel cause this stall even when USB cable is short and of good quality.
3. Move the mains cable away from the USB cable. These are the mains cable for your stepper and spindle.
4. Most important, use a dedicated computer for your CNC. If you can't, just unplug other USB device while cutting.

matrix-wrangler, 2019-09-19:
Thank you @n0th1n6 , 3 and 4 has always been the case that i followed strictly. Got a dedicated laptop on the cnc and a UPS + PSU for the CNC away from the mains to avoid interference. It's an old board that i'm using, i eventually figured out that sparks from brushed spindle (both old and new) is causing a lot of interfere for the machine to stall, as opposed to the cnc completes the job flawlessly when it's not connected to the spindle. Easier option would be to replace the board (woodpecker 3.2), which i did, but i wanted to go a little further to find a solution without a replacement board. I'll certainly try the high quality cable, but the params on choke can also help if it manages to negate the fluctuations caused by spindle.