gnea/grbl #598: Inconsistent behavior between job when switching laser mode

quezadaminter, 2019-02-04:
In relation to issue #406 and while writing a post-processor that will automatically engage/disengage GRBL's laser mode, I ran into what may be a bug. The code below:

sends the $32=1 command to engage the laser mode
moves the head to a safe height
moves the head to a starting position
moves the head to a cutting height
moves the head in a rectangular pattern
moves the head to a safe height and position
sends the $32=0 command to disengage laser mode.

If the program is sent again immediately to repeat the job, it will not start and or finish the job properly and the fault is inconsistent in where the program ends. Sending GRBL a soft reset after this will clear the inconsistency and allow the job to run again. No error is reported or sent back from GRBL indicating that it ran into trouble. Running the code through the GRBL-sim also indicates no issues in the code itself (i.e. no invalid G or M code lines).

Without the $32 commands the code executed repeatably and without issue.

(1001)
G90 G94
G17
G21
;G43.1 Z0
(Enable the laser mode!)
(Using X offset: 0)
(Using Y offset: 0)
(Using Z offset: 0)
$32=1
G28 G91 Z0
G90

(2D Contour5)
S12000 M4
G54
G0 X350.6 Y152.4
Z15
Z0
G1 Z-0.2 F333.3
Y261.462 F5000
G3 X345.6 Y266.462 I-5
G1 X60.8
G3 X55.8 Y261.462 J-5
G1 Y43.338
G3 X60.8 Y38.338 I5
G1 X345.6
G3 X350.6 Y43.338 J5
G1 Y152.4
G0 Z0
G28 G91 Z0
G90
G28 G91 X0 Y0
G90
M5
;G49
(Disable the laser mode!)
$32=0

109JB, 2019-02-04:
Why are you sending $32 during a job run. This is an EEPROM function for persistent settings, not for inclusion in normal G-code runs. Using a $ command interrupts Grbl during the read/write to EEPROM process.  You should never have to use a $ command in a normal program.

To better understand, tell us why you have the $32 within the g-code program and we can tell you how it should be done.

quezadaminter, 2019-02-04:
My goal is to enable/disable the laser mode in GRBL at the post-processor level to perform an automated tool change between a router head and a laser head. I understand that the $ commands interrupt the planner and the execution of the motion commands, this is why I am placing them before and after any motion occurs.

I have a CNC machine with both a router and a laser head. The laser head is located at a known offset from the router head. The work I do is a mixture of router carving and laser engraving. I post process a router tool path into gcode and save it to a file. Then I post process a laser tool path and save it to another file. Then I one file to the machine, wait for the program to complete, switch the laser mode manually and send the next file for the next job. I want to streamline this process in my setup by concatenating the two files and sending it just once. This will allow me to run some of the jobs unsupervised so that the tool change between the router head and the laser head can be automated.

I tagged  #406 here because it came up in my research for the feasibility of attempting this idea. I understand that continuously writing to the EEPROM will wear it out and is not an ideal solution, hopefully #406 will get completed and this won't be a problem. However I am interested on working out this concept of the automated tool change for the router/laser pair.

109JB, 2019-02-04:
Then try a G4Px before any $ commands.  The G4 will insure that the planner is empty before the $ command is sent

quezadaminter, 2019-02-05:
@109JB , thank you very much for the tip. Using the dwell command did exactly what I was hoping. It took me a bit to figure out that GRBL was rejecting the $ command while NOT in IDLE (error:8). It makes sense now that the G4 would allow GRBL to switch to IDLE mode and hence accept the $ command.

As reference for the future, here is the updated gcode from the example above that was the reason for posting the issue:

(1001)
(Enable the laser mode!)
(Using X offset: 0)
(Using Y offset: 0)
(Using Z offset: 0)
G4 P1.0
$32=1

G90 G94
G17
G21
;G43.1 Z0
G28 G91 Z0
G90

(2D Contour5)
S12000 M4
G54
G0 X350.6 Y152.4
Z15
Z0
G1 Z-0.2 F333.3
Y261.462 F5000
G3 X345.6 Y266.462 I-5
G1 X60.8
G3 X55.8 Y261.462 J-5
G1 Y43.338
G3 X60.8 Y38.338 I5
G1 X345.6
G3 X350.6 Y43.338 J5
G1 Y152.4
G0 Z0
G28 G91 Z0
G90
G28 G91 X0 Y0
G90
M5
;G49
(Disable the laser mode!)
G4 P1.0
$32=0

Thank you again.

closed 2019-02-05 by quezadaminter