gnea/grbl #600: 4/5 axis questions

109JB, 2019-02-08:
So I'm working on a 5 axis version of Grbl. A and B and they are working, but there is something I would like to attempt to do. I am not very good at C/C++ programming and would like a little assistance on where to start looking in the code to accomplish this. I just want a tip or two on where to start looking and won't barrage with countless questions. Here is the issue I want to attempt to correct:

The "norm" for rotational axes is that if there is a linear (xyz) axis on the same line as a rotational move (ab) that the feed rate should be calculated based only on the linear axes with the rotational axes following in a coordinated fashion to complete when the linear axes complete. If there is no linear axis in the line of code the motion should be calculated based on the rotational axes. For example, a G91 G1 X1 A1 F1 command should complete in one minute and a G91 G1 A1 F1 should also complete in one minute. The way it is currently working, the G91 G1 X1 A1 F1 results in a move that takes 1.414 minutes because the A axis is included in the calculations. In conjunction with this, feedrate reporting should coincide with the above where the reported feedrate is based on only XYZ if present or on AB if only rotational. Where would I start looking in the code to see about "correcting" this?

MeJasonT, 2019-03-19:
Hi John,
I looked at your question and wondered if the axis commands could all share the same line of code and discovered yes they can.  Thanks for the lesson - good to know.
Whilst clicking around the web i found a faunc pdf which may interest you
[http://heim.ifi.uio.no/matsh/inf4500/cnc/fanuc.pdf](url)
I have just obtained a carousel out of a paint shaking machine which is solidly built and looks like a good contender for converting into 4/5 axis for my machine (adding table rotation type functionality)

like you my C# is limited let alone C
I noticed in the pdf that there is Gcode for calling macros could it be a potential option for your GUI so it could call tool change functions etc. obviously intercepted by the GUI and acted upon not handled by GRBL

looking forward to seeing your progress
all the best

Fusseldieb, 2019-04-09:
[There is a 6-axis GRBL fork available.](https://github.com/fra589/grbl-Mega-5X)

If I were you, I would take the above version.

Neither me, nor you seem to know C++ in a "experienced" level, which might be necessary to get GRBL to operate correctly, so it might be better to just take a already done fork.

Putting more axis into the code won't work so easily and if it does, you'll probably experience stability issues with higher motor frequencies. All axis must be on the same physical Arduino PORT and, and, and...

MeJasonT, 2019-04-09:
In had spotted this GUI and every search i do seems to end up back at this link.
the GUI is written in Python and isn't easy for me to figure out how to run the application.  Any pointers would be brilliant.  I have python installed under Win 7 to run Skeinforge 3D printer GUI etc so my machine should be equipped to run it - do i need to write a batch file or something.

Fusseldieb, 2019-04-09:
> In had spotted this GUI and every search i do seems to end up back at this link.
> the GUI is written in Python and isn't easy for me to figure out how to run the application. Any pointers would be [...]

Which GUI are you talking about?

MeJasonT, 2019-04-09:
cn5X
https://github.com/fra589

Fusseldieb, 2019-04-09:
> cn5X
> https://github.com/fra589

Oh, I don't know that panel. 

I use [CNCJS](https://cnc.js.org/) for my CNC machine. All axis work just fine with it. You can define customized G-Code buttons, and a lot more. It's really powerful and useful.

MeJasonT, 2019-04-09:
How do you get CNCJS to run on win7

Fusseldieb, 2019-04-09:
> How do you get CNCJS to run on win7 w/Chrome

To run it just install [npm](https://nodejs.org/en/) (node will install npm) and then just run in the CMD `npm install cncjs -g`.

Tip: _Everytime you install something with npm that you want to be available globally across all CMDs, use `-g` on the installation. In our case, it's needed._

Then after the install just type `cncjs` in the CMD and it will start the server and give you the IP you should access. From there you can control the whole machine. 

Don't worry, if you close the browser, it will still run, as the server caches your whole g-code file and configurations. It's not running directly in the browser, so a browser crash, for example, won't affect the machine or the currently executing task.

The "right" way to do it would be buying a Raspberry Pi, hooking it up to LAN and running CNCJS from there. So even if your PC is off, you can still access the machine with the Pi's IP with your tablet, notebook, whatever...

109JB, 2019-04-10:
> [There is a 6-axis GRBL fork available.](https://github.com/fra589/grbl-Mega-5X) I tested it on my own CNC, it works just fine.

I finally had a chance to download that one and it does appear to work very well and correctly for the rotational axes.  I will likely use this one for now and examine the changes that he made so that I can port those changes over to the ESP32 version of Grbl that I intend to use. The mega version you linked will work for now but I ultimately need faster step generation and the ESP32 version can achieve that.  Thanks for the link.