gnea/grbl #601: Wiki image low quality / hard to read (Page: Connecting Grbl)

MidnaRinku, 2019-02-08:
Can you please update the image on the "Connecting Grbl" wiki page? The labels on the board don't match the pins and that makes it kinda hard to read. I've made some high quality images in Inkscape, if you want to use that. Cheers!

[image]
[image]
[image]
[image]

109JB, 2019-02-09:
The wiki ages aren't locked from editing, so feel free to update the wiki page yourself.

MidnaRinku, 2019-02-09:
Oh I didn't know that. This was my first post ever on github as well, still have no idea how things work around here. I'd prefer if someone else would put it on the wiki, I don't want to screw things up lol.

closed 2019-02-19 by chamnit