gnea/grbl #602: unused pin

jagerjtan, 2019-02-09:
I see there is an unused pin(A4) if grbl is compile to Arduino UNO. 

Is there any command/method like M42 to modify state of this pin for grbl?

Also, reset pin(A0) and resume pin(A2) are not really necessary in use most of the time, how can I modify them for other uses?

HuubBuis, 2019-02-09:
> I see there is an unused pin(A4) if grbl is compile to Arduino UNO.
Is there any command/method like M42 to modify state of this pin for grbl?
Also, reset pin(A0) and resume pin(A2) are not really necessary in use most of the time, how can I modify them for other uses?

There is no command to change the state of unused pins but you can modify the GRBL code to do what ever you want. There is however very little program/memory space left on the Arduino.  If you use an Arduino Mega, there is plenty program/memory left.

I use the Coolant and Mist pin for these purposes because I don't use Coolant or Mist.

chamnit, 2019-02-19:
There are no current plans for the free pin.

closed 2019-02-19 by chamnit