gnea/grbl #603: New message in report

MeJasonT, 2019-02-10:
What is "PS: YZ" in the reply from GRBL in the status message.
I am parsing Grbls status message so that i can drive an external motor controller, all worked well until a new message started to appear in the string.
I never enabled any such reply and i'm confused how it appeared.
I used Xloader to upload the newest 1.1f hex file to an arduino on my bench and it does not transmit what i thing is a status message to do with the work plane.  When insert the programmed chip into the machine this PS:YZ message appears.
Could it be the limit switches report?
It lies between the feed speed message and the applications message for spindle direction/coolant etc

neilferreri, 2019-02-10:
Are you sure it's not Pn?

Assuming it is, yes, it's the switches. They'll only show they're triggered (probably noise).
There's a section about interfacing with grbl in the wiki.

109JB, 2019-02-10:
See https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface where there is a very descriptive writup of all of the reports that Grbl issues.

MeJasonT, 2019-02-10:
Thanks guys,
I found the report description just after i posted
I forgot i had a board which converts the proximity sensors to inputs for my grbl board.  Unfortunately it only works when the external supply is switched on which also feeds the transformer for the motors.  i was just testing the board without activating the motors - limit switch activation makes sense.
However i will have to check again but im sure i got Pn which would indicate a typo in grbl 1.1f latest version 18 Jul 2017.

MeJasonT, 2019-02-11:
Regards Pn,
I had an Epiphany last night, The difference in prefix is probably how Candle CNC Gui displays the pin status.  Its probably the way Denvi has chosen to display it as apposed to the way grbl transmits it.
Its still a minor pain in the arse as my parsing code is looking for the functions being displayed but when there are limit pins activated the message gets moved over one block.  Luckily my parser cant read it and sends zero which deactivates the spindle anyways.  That's what we call in the trade "A Result".

closed 2019-02-17 by MeJasonT