gnea/grbl #604: Homing for dual stepper configuration

Antiheld86, 2019-02-10:
Hi there,

I build my diy CNC router based on the MPCNC. It works with dual steppers on the X and Y-axis. I am running it with a Ramps 1.4 baord with a stepper driver for every single stepper mototr and every motor has its own end switch. At the moment it runs with marlin firmware and the homing is used to square the axis by hitting the end stops.

I am thinking about switching to grbl because marlin always feels like using a 3d printer.  

I am wondering if my actual setup is paussible with grbl. Is there a feature of setting up X and Y dual axis and giving every stepper mototr its own end swich?

Thanks for your help!

MARIOBASZ, 2019-02-27:
I think it's not a good idea: If you have two motors on the y-axis, and you move only one, detect the limit switch, and then repeat the operation for the other motor, you're forcing it to twist instead of traveling perpendicular

dexy31337, 2019-05-26:
Currently I'm working on a grbl-Mega fork to implement this idea for Y axis.
I've build my CNC based on OpenBuilds OX (XY is belt-driven) with two steppers on Y axis with both drivers' step/dir ports wired together into arduino ports.
The problem is that while machine is powered off it loses perpendicularity between X and Y and I believe I can't fix it mechanically. I want to use special homing procedure to regain perpendicularity.
At the moment I've tweaked code in **stepper.c** to do simultaneous movement for both steppers on different pins (having them on same 2560 port) like this:
```
if (st.counter_y > st.exec_block->step_event_count) {
  #ifdef SEPARATE_Y_AXIS 
    st.step_outbits |= ((1<<Y_STEP_BIT)|(1<<Y2_STEP_BIT));
  #else
    st.step_outbits |= (1<<Y_STEP_BIT);
  #endif
    st.counter_y -= st.exec_block->step_event_count;
  #ifdef SEPARATE_Y_AXIS 
    if (st.exec_block->direction_bits & ((1<<Y_DIRECTION_BIT)|(1<<Y2_DIRECTION_BIT))) { sys_position[Y_AXIS]--; }
  #else
    if (st.exec_block->direction_bits & (1<<Y_DIRECTION_BIT) { sys_position[Y_AXIS]--; }
  #endif
```

Now I'm trying to come out with how to make homing approach phase together for both steppers, but then do locate phase for each motor alternately, depending on which limit switch was triggered (one that was triggered goes last to minimize twist).

Any ideas where I should change/add source code to achieve it?