gnea/grbl #607: CNC2417

notsolowki, 2019-02-13:
im sorry im new at this, will this firmware work with the cnc2417 from ebay. i updated the firmware but the axis dont move. i put the hex file that came with the machine back on the nano and the z axis is backwards and seems to be moving slower??? please help thanks!!

109JB, 2019-02-13:
If your machine came with Grbl 0.9 and you flashed with 1.1, all of the settings parameters will be reset. When you go back to 0.9, the same will happen. It sounds like this may be what has happened. Going from 1.1e to 1.1f to 1.1g will not cause this. Likewise going from 0.9h to 0.9i to 0.9j won't cause settings to be reset. This happens only when crossing from a 0.9 version to a 1.1 version because there were many changes made made between 0.9 and 1.1.

You will need to re-configure Grbl for your machine.  Since you are kind of stuck without proper settings at this point, if I were you I would re-flash with the current V1.1, and configure there. This wiki page describes configuration: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

If you decide to stay with 0.9, much of the configuration is the same but there are some differences. How to configure 0.9 is on this page: https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9

notsolowki, 2019-02-13:
sorry while you were typing i posted an update to my question. you are correct this is the info i was looking for. i used that first link a few minutes ago to get the axis back to the correct direction. then i tried printing something and it seemed like it came out much smaller than before. so now i dont'' know if its printing correctly now or before. what settings should i be looking at to make sure it prints the right dimensions. i would like to use gbrl 1.1 but when i send a command to the motors it just sits and hums.

notsolowki, 2019-02-13:
did they change the pinout between versions? if i press up on the control software it moves the mil head right. if i press down it moves the control head right. any other direction is a quiet hum and the wont move

109JB, 2019-02-13:
I'm not familiar with the CNC 2417 that you have. It is possible that the manufacturer had a custom pin mapping. It can be changed but is a bit more involved. Can you provide a picture of your control board to start?

notsolowki, 2019-02-13:
here are some images of my board.  

[image]
[image]
[image]

notsolowki, 2019-02-13:
also i came across this but im not sure how to use it? https://github.com/zlakomanoff/grbl-cnc-2417

notsolowki, 2019-02-13:
okay i figured out how to use that patch file. now my motors are working. but now im not sure about the stepping distance. whats the right way to set the stepping delay/distance??

109JB, 2019-02-13:
Glad you got to where you are now.  First, are the motors moving in the correct directions?  Once we have this right we will move to the distance settings,

To set directions, first run the machine and determine which if any are moving in the wrong direction.  If your test shows that they are all already moving in the correct directions then skip this whole part about direction setting. If some are wrong, then to avoid confusion you should first set $3=0 to set all the pins for "non-inverted" operation. Then run the machine again and check the directions writing down which ones move correctly and which don't. The ones that are wrong will need to be inverted which is done using $3 setting. Then using the following mask table to determine the setting for $3.

Setting Value | Mask | Invert X | Invert Y | Invert Z
0 | 00000000 | N | N | N
1 | 00000001 | Y | N | N
2 | 00000010 | N | Y | N
3 | 00000011 | Y | Y | N
4 | 00000100 | N | N | Y
5 | 00000101 | Y | N | Y
6 | 00000110 | N | Y | Y
7 | 00000111 | Y | Y | Y

for example to invert Y and Z you would enter $3=5 from the table above.  You can enter this in the Arduino IDE Serial Monitor.

No we can move to distance. There are a couple ways to do this, but what you need to determine is how many step are needed to move 1 mm.  

First way is knowing the microstepping of the drive, and the pitch of the screw that drives the axis.  For example if the drive is set for 10 microstepping then that is 10*200 = 2000 steps per revolution because stepper motors have 200 full steps per rev and the microstepping subdivides each step into X number of microsteps.  Now if the screw moves 2mm/rev or in other words is 2mm pitch, then just take 2000/2=1000 steps/mm. This is what you would enter for that axis. The distance settings are as follows:
$100 = setting for steps/mm for X axis
$101 = y axis
$102 = Z axis

The second way is to determine the current setting by issuing a $$ command into the Serial Monitor, which will list all the settings. Records the $100,101, and 102 settings for x, y and z. Now run the machine and program a move of say 100mm and measure as closely as you can the actual distance the machine moves.  Lets say it moves 50mm when programmed for 100mm. The new setting should then be

New setting = (old Setting) * (distance moved)/(distance programmed). 

You would do this for each axis and then send the appropriate $100, 101, or 102 to Grbl in the Serial monitor.

See if that works and then we can work on other settings.

notsolowki, 2019-02-13:
wow thankyou. i now have all the motors moving in the right directions and im going to try to work on the stepping formula. its probably going to take me some time to figure out the stepping part. i know nothing about stepper motors. do most all stepper motors have 200 steps per rev?. and sorry im not sure what you mean by pitch of the screw. this is my $$ printout 

`$$ < $0=10
$1=25
$2=5
$3=6
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
ok`

109JB, 2019-02-13:
The pitch of the screw is how far the axis will move with one revolution of the screw.
Not all steppers have 200 steps per rev, but this is 99.999% of what is used on these type machines.

closed 2019-02-18 by notsolowki