gnea/grbl #608: grblControl software

notsolowki, 2019-02-18:
im using grblControl 0.8.4. the application that came with my cnc247 machine. is there a better version or better software i should use?

carneeki, 2019-02-18:
https://github.com/gnea/grbl/wiki/Using-Grbl

Has several recommendations.

notsolowki, 2019-02-18:
i have to restart grblControl every time i want to zero out my x/y. slightly unrelated but the problem im having besides that is i used inkscape to make a 40x80mm rectangle then i use gcodetools extension to convert it to gcode but when i open it in grblControl the starting origin is way off of the rectangle so the bed has to travel really far to begin cutting

carneeki, 2019-02-18:
You should be able to set the origin in the control software, or even by issuing G-code yourself. Some reading is needed... Use [this](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands) to find all the commands Grbl can run, and [this](http://linuxcnc.org/docs/html/gcode.html) to learn the details of those commands. There will be a few differences but they'll be documented on the Grbl wiki.

To get the ball rolling, `G10` is one to help you set the WCS (Work-Coordinate-System) which Grbl uses to say where 0,0,0 is on the part. Most GUIs should be able to do this for you (UGS for example, you jog over to where you want the axes to be set up, and can press "Xo" "Yo" and "Zo" to zero out each individual axis.

zeevy, 2019-02-18:
> 
> 
> im using grblControl 0.8.4. the application that came with my cnc247 machine. is there a better version or better software i could use?

If you manage to connect bluetooth or having USB OTG then you can try the mobile application. It has easy interface to zero the machine/homing and set work co-ordinates etc..

https://github.com/zeevy/grblcontroller
https://play.google.com/store/apps/details?id=in.co.gorest.grblcontroller

notsolowki, 2019-02-18:
another problem im having with the control software is ever since i update from 0.9 to 1.1f the control soft dont show the animation of the bit cutting the lines in the visualizer. it shows the spindle spinning but dont show it moving around the plane. also under connection status it says unknown?

closed 2019-02-18 by notsolowki

swarfer, 2019-02-18:
A few years ago I tried Inkscape and just could not get any gcode out of it.
So I recommend Sketchup 2016 + SketchUcam instead (yes, I am biased because I develop SketchUcam :-)
At least with Sketchucam knowing where 0,0 is, or setting it to where you want it, is very easy.
https://www.youtube.com/playlist?list=PLm728rLHCWOw-vRvTJwS_5MmKG1fs4sZ0

As for GUI's, there are many, most of them better that that very old Grlbcontrol.   I use [bCNC](https://github.com/vlachoudis/bCNC/releases) but I do like [Candle ](https://github.com/Denvi/Candle/releases) both for its simplicity and also the way it jogs.  I use bCNC because it supports multiple work offsets correctly which Candle does not, and also plugins can be added to it to generate custom Gcode, and many other features.

For simple jobs I also use http://cam.openbuilds.com to generate Gcode and send it directly to the Openbuilds CONTROL driver.

MeJasonT, 2019-02-18:
GRBL setting $10=3 corrects the visualisation error.
the other issue with WCS looks like your machine is using the negative coordinate system and inkscape is setting the start point as if it were using A4 paper printer or a plotter and setting the home axis to 0,0 in the bottom right corner.

notsolowki, 2019-02-18:
> GRBL setting $10=3 corrects the visualisation error.
> the other issue with WCS looks like your machine is using the negative coordinate system and inkscape is setting the start point as if it were using A4 paper printer or a plotter and setting the home axis to 0,0 in the bottom right corner.

as a work around i just moved the project in inkscape closer to the origin. and the tutorial i was watching choose cylinder in the gcodetools  "tools library"  selecting default seems to clear up a little bit. but im still not sure how to properly set 0,0 so when i open it in grblControl it dont start the mill toooo far away from what im cutting.