gnea/grbl #609: Visualization not following lines in control software

notsolowki, 2019-02-18:
before i upgraded from 0.9j to 1.1f i could watch the spindle location change in the grblControl software. despite most everything else working as it should i also i get "unknown" in the status window?

notsolowki, 2019-02-18:
i have tried downgrading back to pre 1.1 and they all work properly with the software. its as soon as i go to any version of 1.1 and i loose communication with the software. come to find out i can also zero out x/y with version 0.9 dont seem like it works properly after upgrading.

closed 2019-02-18 by notsolowki

neilferreri, 2019-02-18:
Use a different UI. When was the last time grblControl was updated?
I like CNCjs.

MeJasonT, 2019-02-18:
Change $10=3 in GRBL

notsolowki, 2019-02-18:
@MeJasonT yea i end up figuring that out. its my own fault for not reading closer. there it was  status report. im not sure whats so bad about grblControl it seems simple enough, it seems as if there is a lot of software out there. i tried cBNC i just it might be overkill for my little engraving machine. i just upgraded the motor to a 400w brush-less  motor and controller.

langwadt, 2019-02-18:
which version of grblcontrol are you using, does it even understand grbl 1.1 ?

papillion47, 2024-09-26:
Benim GRBL programımda da benzer sorun var. Program açıldığında "visualizer" yani görselleştirici penceresi de açılır ve yapılacak iş ve freze çakısının hareketleri oradan takip edilir, ilk ayarlar buna göre yapılırdı. Bu şekilde aylarca güzel çalıştı. Sonra 1-2 ay ara verip yeniden çalıştırdığımda "visualizer" boş olarak karşıma çıktı. Neyin sebep olduğunu bir türlü keşfedemedim. Programı silip yeniden yüklemek de işe yaramadı. Program herhangi bir hata mesajı vermiyor. Diğer her şey doğru çalışıyor. Yardımlarınızı bekliyorum. Teşekkürler.. Yardım için [papilllion4@gmail.com](mailto:papilllion4@gmail.com)