gnea/grbl #611: Design Software

notsolowki, 2019-02-19:
im trying to figure out what software to use for my cnc2417 mill. right now im using inkscape and the gcode export plugin. inkscape is giving me trouble trying to define the origin. it seems like either the software dont work right or cant give me the proper gcode. can someone recommend a free software to use to design the gcode

notsolowki, 2019-02-19:
in the control software my machine moves the proper direction when jogging but if i use inkscape to make a gcode file using gcodetools extension if cuts y- and y+ backwards.

n0th1n6, 2019-02-19:
I don't think this is even a grbl issue.

wlaw, 2019-02-19:
What are you trying to make?

That is very important on the list.

Also, you can try many of the programs out before you buy.

But a good starter program is CamBam.

It is easy to use and a good starter. Also when you decide to purchase it's
149.00

Nothing good is free as far as CNC software.

<https://github.com/gnea/grbl/issues/611#issuecomment-464934282>, or mute
the thread

notsolowki, 2019-02-19:
im just trying to print a 84x54 rectangle. it has the dimensions right but it cuts everything mirrored.

carneeki, 2019-02-19:
If I understand correctly, the Y axis just needs to be swapped?
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

The $3 setting lets you change the direction. If you need help figuring this out, can you paste the current $3 setting to us?

notsolowki, 2019-02-19:
well when i jog y- in the control software the bed some come towards me right? also heres the config

$$ < $0=10
$1=25
$2=0
$3=6
$4=0
$5=0
$6=0
$10=2
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=400.000
$101=400.000
$102=400.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000

chamnit, 2019-02-19:
Not a Grbl issue. Closing.

closed 2019-02-19 by chamnit

carneeki, 2019-02-19:
Im not sure how your machine is supposed to operate with Y+ and Y-, but if all you want to do is swap the Y axis, given what you pasted we can calculate the mask. Masks are calculated by adding like this:
```
Z Y X 
4+2+1
```
Right now, you have `$3=6` Which is `4+2` ( `Z+Y` ).
To swap we do `Z+0+0 = 4+0+0 = 4`.
So setting `$3=4` should swap the Y axis given what you pasted.

Good luck, and always check the manual to be sure you're doing the right thing. The developers put a lot of effort into it!

notsolowki, 2019-02-19:
well the cnc2417 machine i bought didnt come with a manula and the software is pretty plain. now im not sure if my motors are moving in the right direction. maybe i do have y wrong? i figured in grblControl y- mean move toward negative side of y plane  which should move the mill bed to me if im standing infront of it. i dont know where else to ask i feel bad asking all this here sorry

notsolowki, 2019-02-19:
i made files with cambam, inkscape, adobe illustrator, they all came out print the y axis backwards it seemed. how do i get to the bottom of this ?

carneeki, 2019-02-19:
If everything is producing a mirror image, then perhaps the Y axis is configured backwards. It might not be, but if everything has done it backwards so far, it is easy to change $3 and easy to change it back.

Another option is to use Inkscape / Illustrator / Cambam to "vertically flip" the image every time as soon as you open it, but this will get annoying.

MeJasonT, 2019-03-05:
WCS, you are trying to produce a cam file from a vector program.  Inkscape is cutting as if the work is on A4 paper ie home is bottom left and cnc machines and grbl by default use top right as homing position (negative coordinate system). you can invert your axis in grbl which will provide an A4 type of layout but better to work in proper machine coordinates as your file will be easier to share and transmit to others AND run on pro machines.  There must be a solution where inkscape has been converted to run on negative coordinate system.